PX4/PX4-Autopilot #2275: If the mission does get rejected initially, the fixed mission is still never executed

LorenzMeier, 2015-06-04:
Try by setting the first waypoint too far initially, then move it closer. It still will never accept it.

tumbili, 2015-06-05:
@LorenzMeier I rather have the problem that it always accepts it.

tumbili, 2015-06-05:
@LorenzMeier Do you agree that we should move the check to the mission feasibility check?

LorenzMeier, 2015-06-05:
Yes, that makes sense.

tumbili, 2015-06-05:
@LorenzMeier Actually not, because the home distance is not know a priori.

tumbili, 2015-06-05:
@LorenzMeier I'm just rebuilding QGC, it won't let me send commands.

tumbili, 2015-06-05:
But I tested on the other pc and it seems to work there. Couldn't yet reproduce the problem.

tumbili, 2015-06-05:
Works for me. I first set the waypoint far away and it complained. Moving the waypoint close enough made it accept the mission.

LorenzMeier, 2015-06-05:
Was it already flying then?

tumbili, 2015-06-05:
No. There is no difference between fw and mc right?

tumbili, 2015-06-05:
I'm testing mc with jMAVSim.

tumbili, 2015-06-05:
What were you testing on?

LorenzMeier, 2015-06-05:
HIL on fixed wing. I can't reproduce either right now, so closing this. Sorry for the confusion!

LorenzMeier, 2015-06-05:
Ok, got it. This is after a jump in position in HIL. Looks like the auto mode is suffering from some sort of NaN state that never gets reset.

tumbili, 2015-06-05:
@LorenzMeier What do you mean by jump in position?

LorenzMeier, 2015-06-06:
I traced this back to my HIL setup generating NaN states in the system. Need a better detector: https://github.com/PX4/Firmware/issues/2291

closed 2015-06-06 by LorenzMeier