PX4/PX4-Autopilot #2276: After the last mission item the system is not flicking into loiter, but just continues forever

LorenzMeier, 2015-06-04:
Try by executing a short mission, it will continue on the course of the last waypoint.

tumbili, 2015-06-05:
Trying this one now with fw

tumbili, 2015-06-05:
@LorenzMeier Also can't reproduce this one, plane is circling nicely around last waypoint.

kd0aij, 2015-06-05:
Tried my first auto WP mission yesterday using QGC to set waypoints at the field, instead of preprogramming them at home. Altitude setting worked correctly this time, but since I had only partial map data (my phone battery runs down before the QGC map fills in, and QGC doesn't seem to do any caching) I had to place the WPs "in the blind". And I didn't know whether to expect a loiter at the 2nd (last) WP. Apart from a rather violent stop at the first WP, this went OK, but it turns out I switched out of auto mode too soon to test this bug. Will try again in the next weather window.

BTW, are WPs logged? We need to add WPs to the KML track.\

http://dash.oznet.ch/view/bonCQfoetE6NU8HvYucadY
[image]

LorenzMeier, 2015-06-06:
Thanks Mark! I traced this back to my HIL setup generating NaN states in the system. Need a better detector: https://github.com/PX4/Firmware/issues/2291

closed 2015-06-06 by LorenzMeier

kd0aij, 2015-06-07:
@LorenzMeier I just noticed that the Q attitude estimator is not an option for FW platforms. Is it not compatible with the FW apps?

LorenzMeier, 2015-06-07:
Right now its not. It would need additional interfacing, e.g. airspeed to the inav estimator first.