PX4/PX4-Autopilot #2279: Limit pitch / roll target in manual mode when hitting mixer limits

LorenzMeier, 2015-06-04:
Currently we can have the situation that the mixer goes into failsafe mode and gives up on yaw when exceeding its possible spread. While that is the right fail-safe strategy, its not great for manual acrobatic flying.

LorenzMeier, 2015-06-04:
@tumbili This here does not look correct - the comment says it scales yaw, but in fact the math there replaces the commanded yaw entirely, not taking the original yaw demand into account:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp#L306-L320

We probably should also retain some yaw demand, e.g. scale down the demand up until some threshold, e.g. 0.1 or 0.05. Anything below that threshold still gets mixed even when the system is approaching the limits. The idea is to maintain the yaw heading with some minimal contribution, but not to allow heading changes any more.

This should fix the usage problems @AndreasAntener was describing.

tumbili, 2015-06-04:
@LorenzMeier No I don't think so. Just above you see that yaw is mixed in. Now if we see that the mixing in of yaw caused our output to violate bounds we just just the yaw which will lead to the output being on the bound. You don't need to yaw command anymore for that.

LorenzMeier, 2015-06-04:
So in other words it subtracts a scaled yaw? In that case we would just need to limit the subtracted quantity so that it doesn't subtract so much that we loose all yaw authority. I have no idea about the right magic number for this though, nor do I have a better / solid idea not involving magic numbers.

tumbili, 2015-06-04:
The math is like this (I left away the scales):

out = R + P + thrust + Y -> e.g out is negative now

we want out to be exactly 0 not negative, so we set it to 0 and solve for Yaw:

Y = - (R + P + thrust)

tumbili, 2015-06-04:
I think for full thrust the case is clear: we have to dedicate some thrust to yaw control, meaning that if the user wants to go full thrust up and yawing at the same time he will have to give some thrust away to yawing. But for the rest I would say we should leave it as is.

LorenzMeier, 2015-06-04:
The problem is that users who fly full throttle and pitch forward loose yaw control. That's absolutely deadly during manual flying, so we have to fix that.

tumbili, 2015-06-08:
@LorenzMeier I just did I flight where I allow to give 0.15 thrust away to yaw. Was able to do full thrust pitch forward, yaw tracked well. However, to change yaw during full thrust is not possible since 15 % is too little.

tumbili, 2015-06-08:
Here is the log  http://dash.oznet.ch/view/devkueALTqXyBCghqhueyV

tumbili, 2015-06-08:
@LorenzMeier If you use flightplot and look at the motor outputs at 233s you can see very nicely what the mixer does.

LorenzMeier, 2015-06-08:
Is it not possible to change yaw or does it loose the yaw tracking? I think no change is fine, but it shouldn't loose the yaw angle.

tumbili, 2015-06-08:
@LorenzMeier It does not loose it (although the tracking could be a bit better). I'm going to  play with this param. In one extreme case I will set it high so if you apply full thrust and yaw you should get a low thrust value and see it yawing (commanded). We will have to agree on a value which makes sense.

tumbili, 2015-06-08:
[image]

tumbili, 2015-06-08:
Notice the spike in the motor outputs at 233s

LorenzMeier, 2015-06-08:
@tumbili Where is the branch / PR for this?

tumbili, 2015-06-08:
@LorenzMeier Oh, forgot to push it. Just going for another test flight (removing position controller carrot approach and then will push everything)

LorenzMeier, 2015-06-08:
Addressed in b571d704883d850242b879d02b86fcb6907941e6.

LorenzMeier, 2015-06-12:
@AndreasAntener Reopening, please put the log in here.

AndreasAntener, 2015-06-12:
These logs should show it:
http://dash.oznet.ch/view/M8SyLmxiT7hmvJ3ZaGXRnD
http://dash.oznet.ch/view/uZ9XE9DKtGRJzvvaeeVfaF

LorenzMeier, 2015-06-12:
http://dash.oznet.ch/view/M8SyLmxiT7hmvJ3ZaGXRnD#AS_Yaw_PLOT Does only show minor yaw spikes - is this what you're referring to? Was this with off board control?

AndreasAntener, 2015-06-12:
This was all manual. Judging from the ground, the yaw error was about 15-20 degrees before we let go of pitch. I'm pretty sure that it happened in those flights. If there's time maybe we can to a dedicated test for that today.

LorenzMeier, 2015-06-12:
I think you want this fix for off board anyway:
https://github.com/PX4/Firmware/pull/2341

If you test it we can merge it relatively quickly.

The manual throttle target should have been limited to 90%, so I need to check why its not.

LorenzMeier, 2015-06-12:
The issue was the thrust compensation when pitched. We already had a limit on the other end, and I've just updated the default param value:
https://github.com/PX4/Firmware/pull/2343

It would be great if you could test that one.

AndreasAntener, 2015-06-12:
Ah, thanks. I guess then we flown with max thrust 1 again.

AndreasAntener, 2015-06-12:
Looks good now, thanks :)

PR #2343 (2015-06-12): Update MC thrust limit default param value and add explanation
This should fix #2279 for good.
closed 2015-06-12 by AndreasAntener