PX4/PX4-Autopilot #2281: Takeoff detection on fixed wing: Make detector more robust

LorenzMeier, 2015-06-04:
The current approach can lead to a flurry of false detections:

```
[01:22:03.279 - COMP:50] Critical: TAKEOFF DETECTED
[01:22:05.630 - COMP:50] Critical: LANDING DETECTED
[01:22:39.493 - COMP:50] Critical: TAKEOFF DETECTED
[01:22:41.953 - COMP:50] Critical: LANDING DETECTED
[01:23:02.167 - COMP:50] Critical: TAKEOFF DETECTED
[01:23:05.355 - COMP:50] Critical: LANDING DETECTED
[01:23:08.819 - COMP:50] Critical: TAKEOFF DETECTED
[01:23:12.070 - COMP:50] Critical: LANDING DETECTED
```

LorenzMeier, 2015-06-06:
Fixed on release branch.

closed 2015-06-06 by LorenzMeier

tnedev, 2015-07-30:
It seems that I no longer have the landing detection working properly (there is no landing detected message).
It also looks like this may be the cause of a regular propeller breaking on auto landing. When landed the motor keeps trying to spin (could be ESC fault also).  
I would record this in a MAVLink log and get back with data.

LorenzMeier, 2015-07-31:
The land detector is not suitable for prop stopping on auto landing. You need to set the min throttle on landing for that in the parameters to zero.

LorenzMeier, 2015-07-31:
https://pixhawk.org/users/fixedwing_autoland

flyver, 2015-07-31:
The min throttle for landing is already 0. 
Few times when I left the craft land in auto and stay in auto until I went to take it found the motor still trying to spin on the ground. Since then when the craft lands I immediately switch to manual but that is not ok. Any other reasons which may cause it?

tnedev, 2015-07-31:
The min throttle for landing is already 0. 
Few times when I left the craft land in auto and stay in auto until I went to take it found the motor still trying to spin on the ground. Since then when the craft lands I immediately switch to manual but that is not ok. Any other reasons which may cause it?

LorenzMeier, 2015-07-31:
You need to set the TLALT accordingly as well.