PX4/PX4-Autopilot #2284: Question about MC_POS_CONTROL

Royalcheese, 2015-06-05:
Hello everyone,
I'm working with px4 on a quadcopter, with ROS, MAVROS, ODROID, OPTITRACK, OFFBOARD and indoor.
For simulation, i'm sending a stream of setpoint (x,y,z)=(0,0,0.5), and a mocap position (x,y,z) = (0,0,cos(t)) ;  

I would like to understand how does work mc_pos_control in order to understand thrust output we can get on QGC. I'm using raw code and the paper "minimum snap trajectory generation and control for quad-rotor", in order to understand.
From what I understood , thrust = F(velocity) , F being roughly a PID regulator.
but there is several points confusing me : 
1/
'''math::Vector<3> pos_err = _pos_sp - _pos;

				_vel_sp = pos_err.emult(_params.pos_p) + _vel_ff; '''

A velocity added to a position ? That not homogeneous , I don't understand.
2/
Lowering z_p parameter ,my simulation on drone gives this (failures are due to overload of communication protocol) : 
[image]

Highering this parameter give this : 
[image]

I really don't understand how can this parameter inducing a phase shift....

3/ I tried to do an (ugly) scilab simulation of mc_pos_control . The result is here : 
[image]

I know that it doesn't represent mc_pos_control at all , but it is clear for me that the 2/ problem appears here. 

If anyone could clarify those point , I would be grateful . 

by the way , '''.emult ''' is just a function which take param and multiply the variable with ? 

Thanks in advance for answers !
have a nice day

kd0aij, 2015-06-05:
just in answer to your question about "emult": right clicking on that name in Eclipse and choosing "open declaration" takes you to line 286 of Vector.hpp in src/lib/mathlib/math.

tumbili, 2015-06-05:
@Royalcheese Hi
_vel_sp = pos_err.emult(_params.pos_p) + _vel_ff;

This is a cascaded control system. The outer loop computes position error based on estimator position and desired position. The error of the outer loop gives  you the setpoint for the inner velocity loop. You can think of this "pos_err.emult(_params.pos_p)" as taking a derivative of a position value which will give you velocity. Then again for the velocity error you compute similarly a desired acceleration which finally has the dimension of your inputs (motor thrust) and you can apply it.

Royalcheese, 2015-06-08:
Ok, first thank you a lot for answers  @tumbili  @kd0aij  !
So pos_p parameter can be assumed as  a vector of s-1 dimension ? 
I think I understand better understand how it does work now thank you !

Royalcheese, 2015-06-09:
Sorry for modifying the precedent comment

LorenzMeier, 2015-06-14:
@Royalcheese Please defer the further discussion to the mailing list at https://groups.google.com/forum/#!forum/px4users. We're reserving this system to discuss issues found with the system, not general questions. Thanks!

closed 2015-06-14 by LorenzMeier