PX4/PX4-Autopilot #2285: Selecting a HIL airframe via QGC results in inability to arm

kd0aij, 2015-06-05:
Even though the commander module appears not to require success of the sensor preflight check, one is forced to perform sensor calibration in order to arm.

LorenzMeier, 2015-06-05:
I can't confirm this - I'm arming all the time with a failing prearm check.

kd0aij, 2015-06-05:
Strange. Here's what my console shows when I try to arm from QGC's HIL screen:

```
telem> PREFLIGHT FAIL: MAG #0 SELFTEST FAILED                                   
telem> PREFLIGHT FAIL: ACCEL #0 SELFTEST FAILED                                 
telem> PREFLIGHT FAIL: ACCEL #1 SELFTEST FAILED                                 
telem> PREFLIGHT FAIL: GYRO #0 SELFTEST FAILED                                  
telem> PREFLIGHT FAIL: GYRO #1 SELFTEST FAILED                                  
telem> Preflight check failed, refusing to arm                                  
commander: INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_ARMED               
```

and QGC says it's disarmed... I assume QGC shows your system armed.

LorenzMeier, 2015-06-05:
Have you started the HIL mode at this point? Does it report being in HIL?

kd0aij, 2015-06-05:
```
NuttShell (NSH)                                                                 
nsh> navigator: offboard mission updated: dataman_id=0, count=0, current_seq=0  
sdlog2: log buffer size: 22528 bytes                                            
telem> settings autosaved                                                       
mavlink_if0: Disabling hardware flow control                                    
sercon: Registering CDC/ACM serial driver                                       
sercon: Successfully registered the CDC/ACM serial driver                       
[i] microSD mounted: /fs/microsd                                                
nshterm [6:70]                                                                  
param: selected parameter default file /fs/mtd_params                           
[param] Loaded: /fs/mtd_params                                                  
rgbled: no RGB led on bus #2                                                    
blinkm: no BlinkM found                                                         
nsh: pca8574: command not found                                                 
 1: + SYS_AUTOCONFIG: match                                                     
 1:   SYS_USE_IO: match                                                         
 1001: + SYS_AUTOSTART: match                                                   
  MC_ROLL_P: curr: 6.0000 -> new: 7.0000                                        
  MC_ROLLRATE_P: unchanged                                                      
  MC_ROLLRATE_I: unchanged                                                      
  MC_ROLLRATE_D: curr: 0.0020 -> new: 0.0030                                    
  MC_PITCH_P: curr: 6.0000 -> new: 7.0000                                       
  MC_PITCHRATE_P: unchanged                                                     
  MC_PITCHRATE_I: unchanged                                                     
  MC_PITCHRATE_D: curr: 0.0020 -> new: 0.0030                                   
  MC_YAW_P: curr: 2.0000 -> new: 2.8000                                         
  MC_YAWRATE_P: curr: 0.3000 -> new: 0.2000                                     
  MC_YAWRATE_I: curr: 0.0000 -> new: 0.1000                                     
  MC_YAWRATE_D: unchanged                                                       
  MC_YAW_FF: unchanged                                                          
  MPC_THR_MAX: unchanged                                                        
  MPC_THR_MIN: unchanged                                                        
  MPC_XY_P: unchanged                                                           
  MPC_XY_VEL_P: unchanged                                                       
  MPC_XY_VEL_I: unchanged                                                       
  MPC_XY_VEL_D: unchanged                                                       
  MPC_XY_VEL_MAX: unchanged                                                     
  MPC_XY_FF: unchanged                                                          
  MPC_Z_P: unchanged                                                            
  MPC_Z_VEL_P: unchanged                                                        
  MPC_Z_VEL_I: unchanged                                                        
  MPC_Z_VEL_D: unchanged                                                        
  MPC_Z_VEL_MAX: curr: 5.0000 -> new: 3.0000                                    
  MPC_Z_FF: unchanged                                                           
  MPC_TILTMAX_AIR: unchanged                                                    
  MPC_TILTMAX_LND: unchanged                                                    
  MPC_LAND_SPEED: unchanged                                                     
  PE_VELNE_NOISE: curr: 0.3000 -> new: 0.5000                                   
  PE_VELD_NOISE: curr: 0.5000 -> new: 0.7000                                    
  PE_POSNE_NOISE: unchanged                                                     
  PE_POSD_NOISE: curr: 0.5000 -> new: 1.0000                                    
  PE_GBIAS_PNOISE: unchanged                                                    
  PE_ABIAS_PNOISE: curr: 0.0002 -> new: 0.0001                                  
  NAV_ACC_RAD: curr: 10.00000 -> new: 2.0000                                    
  RTL_RETURN_ALT: curr: 60.00000 -> new: 30.00000                               
  RTL_DESCEND_ALT: curr: 30.00000 -> new: 10.00000                              
+ SYS_AUTOCONFIG: curr: 1 -> new: 0                                             
px4io: CRCs match                                                               
dataman: Unknown restart, data manager file '/fs/microsd/dataman' size is 10150s
[MS5611_SPI] on SPI bus 4 at 2 (11000 KHz)                                      
ms5611: no device on bus 4                                                      
[MS5611_SPI] on SPI bus 1 at 3 (11000 KHz)                                      
ms5611: no device on bus 1                                                      
ms5611: no device on bus 2                                                      
<adc> init done                                                                 
ver hwcmp match: PX4FMU_V2                                                      
hmc5883: no device on bus 2                                                     
[... cut ...]
  146 240 FIFO TASK      WAITSEM  px4io()
  179 250 FIFO TASK      WAITSEM  position_estimator_inav()
  116 100 FIFO TASK      WAITSEM  dataman()
  182 250 FIFO TASK      WAITSEM  mc_att_control()
  152 100 FIFO TASK      WAITSIG  mavlink_if0(10004710, 1000ad60, 100082e0, 100)
  153 175 FIFO PTHREAD   WAITSEM  mavlink_rcv_if0(20011b40)
  186 250 FIFO TASK      WAITSEM  mc_pos_control()
  188 100 FIFO TASK      WAITSIG  land_detector()
  190 120 FIFO TASK      WAITSEM  navigator()
          
```

kd0aij, 2015-06-05:
After pressing the Arm button:
[image]

LorenzMeier, 2015-06-05:
You're not in HIL mode. You also can't switch to HIL if you already armed. So the issue is really that you need first to enter HIL mode, then try to arm. The safety checks wouldn't you let perform the steps in the order you're trying anyway.

LorenzMeier, 2015-06-05:
And the arm button doesn't exist ;-). Its the safety button, and you need to arm after you disengaged the safety.

kd0aij, 2015-06-05:
I meant the "Arm System" button, which is what I meant by "arm from QGC's HIL screen"

kd0aij, 2015-06-05:
and I've been using HIL for more than a year now...

LorenzMeier, 2015-06-05:
You need to be in HIL, and you're not on your screenshot - could you show the output of the system when you're in HIL and try to arm?
I just killed my sensor calibration to test this and could arm.

kd0aij, 2015-06-05:
please explain how I would "get into HIL mode"

LorenzMeier, 2015-06-05:
Connect to the simulator, which will trigger the state switch to HIL. I'm not sure why you would try to arm the system without being connected to the simulator and having the simulation state loaded in the system?

kd0aij, 2015-06-05:
Right. Yesterday, I was following the normal procedure of starting Xplane9, connecting from QGC, then attempting to arm the system. Today I tried to shortcut the process, since I thought that arming was independent of the simulator. And, of course, all those failure messages give a strong expectation that something is wrong.

This time, after restarting everything, QGC crashed when I tried to connect to Xplane9.

kd0aij, 2015-06-05:
OK, on the second complete restart everything went smoothly, except for the preflight fail messages on the console. Perhaps this was all due to problems with the connection from QGC to sim, or a perception that the failure messages were showstoppers. 

Thanks for the enlightenment :)

closed 2015-06-05 by kd0aij

LorenzMeier, 2015-06-05:
I've filed this: https://github.com/mavlink/qgroundcontrol/issues/1644 I will also check if I can enable the preflight circuit breakers for the HIL config to not confuse developers.