PX4/PX4-Autopilot #2294: Test all compilers in Travis which are allowed in the makefile

LorenzMeier, 2015-06-06:
Currently we're not testing all supported compilers in Travis. Current issue to be addressed:

```
PX4/Firmware/src/lib/uavcan/libuavcan/include/uavcan/marshal/array.hpp:516:13: error: comparison is always false due to limited range of data type [-Werror=type-limits]
```

LorenzMeier, 2015-06-06:
@pavel-kirienko @davids5 It would be great if you could have a look at the error message above.

pavel-kirienko, 2015-06-06:
What compiler produced this error?

kd0aij, 2015-06-06:
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20140401 (release) [ARM/embedded-4_7-branch revision 209195]

pavel-kirienko, 2015-06-06:
Fixed in master.

dagar, 2016-02-28:
Do you want to support gcc 4.7-4.9? We could easily split this into 3 travis-ci linux builds that run in parallel, but we'd need to decide which one deploys to s3 and github releases.

LorenzMeier, 2016-02-28:
4.8 and 4.9 would be good.

LorenzMeier, 2016-03-24:
Doing that today, thanks @dagar

closed 2016-03-24 by LorenzMeier