PX4/PX4-Autopilot #2297: No LIDAR Lite data through `DISTANCE_SENSOR` msg in QGC

TSC21, 2015-06-07:
Guys I'm not being able to get LIDAR Lite data through `DISTANCE_SENSOR` message in QGC in any way. I confirmed that the driver is working and publishing data, but can't get that data to be received by Mavlink msg in QGC. From PX4Flow sonar, in the other hand, I can receive `DISTANCE_SENSOR` msg with data when this one is connected.

Can anyone reproduce this problem or give a solution to this? I reviewed the code and it seems to be ok, but two or more pairs of eyes looking is always better. Tried with [master](https://github.com/PX4/Firmware) branch and [jgoppert:cart_pos_estimator](https://github.com/jgoppert/Firmware/tree/cart_pos_estimator) branch, from which [this](https://github.com/PX4/Firmware/pull/2096) PR is being made.

mhkabir, 2015-06-07:
Yes. I have noticed this from the beginning. 

When two at connected, only data from the sensor which bags the topic is coming through (i.e The driver which starts first) 

We need a multi-sub in the mavlink message with subscription handles to 2 instance IDs for this to work, I think. (we should see alternating messages in QGC then)

mhkabir, 2015-06-07:
By default, the normal orb_subscribe sets instance ID to zero, iirc, hence the Lidar data shouldn't be coming through in QGC even when its is only connected.

I'm unable to test the above hypothesis since my wiring is deep routed and I cannot  disconnect the Flow module.

Edit :  I *could* disable the flow driver startup which should prove my guess.

TSC21, 2015-06-07:
@mhkabir the problem is not just that. I'm aware that multi-sub is not working but the thing here is just that LIDARLite is now the only ranger I have connected and I can't get a `DISTANCE_SENSOR` msg received on QGC. Don't know what is happening, cause I think it was working in the first driver+`DISTANCE_SENSOR` support implementation.

TSC21, 2015-06-07:
> By default, the normal orb_subscribe sets instance ID to zero, iirc, hence the Lidar data shouldn't be coming through in QGC even when its is only connected.

Since it is the only one connected, it should work since when I have only LIDAR Lite connected I get `rangefinder0` when issuing `ls /dev`.

mhkabir, 2015-06-07:
I'm pretty sure it wasn't. The instance ID is set to 1 in the lidar driver, which mavlink sender shouldn't even subscribe to, the way its done currently.

TSC21, 2015-06-07:
I tried some other stuff like setting an instance here: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2219 but it didn't help.

TSC21, 2015-06-07:
Also, in the new implementation of @jgoppert PR of LPE, I set https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_messages.cpp#L2234 to `msg.id = dist_sensor.id;`

TSC21, 2015-06-07:
>  The instance ID is set to 1 in the lidar driver

Are you sure? Where is that defined in the default code?

mhkabir, 2015-06-07:
Oh wait. Hmm. When I say ID I mean the ID which uORB assigns to the topic. 
Then it should be according to order the drivers start up. Instance ID in the topic is different from what I mean.

jgoppert, 2015-06-09:
@TSC21 try with latest LPE, I have it working.

TSC21, 2015-06-09:
@jgoppert you got msg in QGC with LIDAR data?

jgoppert, 2015-06-10:
Yes.

TSC21, 2015-06-10:
Hmm ok I will test when possible. What did you change so it could work?

jgoppert, 2015-06-10:
I set the subscription identifier to 1 in LPE. Also make sure you have the servo rail powered with a battery for the lidar.

TSC21, 2015-06-10:
> Also make sure you have the servo rail powered with a battery for the lidar.

Yes that I always had.

> I set the subscription identifier to 1 in LPE.

You mean that setting a subscription in a module sets the Mavlink stream? That's new to me xD Will try when I have the vehicle with me.

jgoppert, 2015-06-10:
In the driver it was set to instance 1, so had to set the instance to 1 in
the sub. But now after syncing with master, had to set it back to 0. Never
had a problem seing it in qgc though.

TSC21, 2015-06-10:
hmm.. what do you have in your `rc.usb` and `extras.txt` files?

jgoppert, 2015-06-10:
I'm not usnig an extras.txt file. I'm using the ll40ls enable flag.

bansiesta, 2015-06-10:
Ok, I looked into this. I tried starting the ll40ls first with the param and second with the extras.txt:

**param: SENS_EN_LL40LS 1**

```ll40ls start pwm``` happens before mavlink is started and ll40ls gets the ORB class instance 0.
Mavlink subscribes to instance 0 and the message arrives in QGC.

**extras.txt**

The file extras.txt is called pretty much at the end of the startup, so after mavlink. This means mavlink creates the distance_sensor topic with instance 0. When the ll40ls finally starts up it sees the topic as already used/blocked and creates another one, so ORB class instance 1.
This means 0 does never get updated and mavlink/QGC does not get distance_sensor updates.

I think this is a flaw of the uORB multi pubsub that needs to be fixed. For now, the advice is to use the SENS_EN_LL40LS parameter instead of the extras.txt.

mhkabir, 2015-06-11:
Yes, this is exactly what I observed. The uORB instance ID gets assigned in the order of driver startup.

For me, this complicates things even more, since the flow driver also bags a topic instance.

bansiesta, 2015-06-11:
@mhkabir Yes, but I don't actually worry about Flow vs. Lidarlite right now but rather about mavlink vs. Lidarlite...

TSC21, 2015-06-11:
@LorenzMeier ping.

LorenzMeier, 2015-06-25:
This should be fixed on master now - can you re-try?

AndreasAntener, 2015-10-29:
I believe this is fixed.

closed 2015-10-29 by AndreasAntener