PX4/PX4-Autopilot #2298: what to do for nonlinear actuators

taogashi, 2015-06-07:
Hi all, I think the current mixer in PX4 handles linear or quasi-linear actuator only, what if those actuators with strong nonlinearity?  If I implement a nonlinear function to map the mixer's output, where should I apply it? Thanks.

tumbili, 2015-06-08:
@taogashi The mixer is define here https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp
Probably the easiest way would be to modify the desired moments and thrust for the axes. Then you don't need to change the entire mixer.
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/mixer/mixer_multirotor.cpp#L214

taogashi, 2015-06-09:
Thanks @tumbili , it seems to be a clean solution.

taogashi, 2015-06-17:
Hi @tumbili, I'm not very clear how the function  `load_mixer_file` behaves.
I suppose it is called when we run `mixer load <dev> <mixer file>`, as in src/systemcmds/mixer/mixer.cpp line 105:

```cpp
static void
load(const char *devname, const char *fname)
{
        int             dev;
        char            buf[2048];

        /* open the device */
        if ((dev = open(devname, 0)) < 0)
                err(1, "can't open %s\n", devname);

        /* reset mixers on the device */
        if (ioctl(dev, MIXERIOCRESET, 0))
                err(1, "can't reset mixers on %s", devname);

        if (load_mixer_file(fname, &buf[0], sizeof(buf)) < 0)
                err(1, "can't load mixer: %s", fname);

        /* XXX pass the buffer to the device */
        int ret = ioctl(dev, MIXERIOCLOADBUF, (unsigned long)buf);

        if (ret < 0)
                err(1, "error loading mixers from %s", fname);
        exit(0);
}
```

However, I modified the function `load_mixer_file` by commenting most of its code, then it looks like:

```cpp
int load_mixer_file(const char *fname, char *buf, unsigned maxlen)
{
        FILE            *fp;
        char            line[120];

        /* open the mixer definition file */
        fp = fopen(fname, "r");
        if (fp == NULL) {
                warnx("file not found");
                return -1;
        }

        /* read valid lines from the file into a buffer */
        buf[0] = '\0';
        for (;;) {
                //DEBUG
                if (NULL == fgets(line, sizeof(line), fp))
                        break;
                printf("%s", line);
        }

        fclose(fp);
        return 0;
}
```

mean to see all the content in my mixer file, which is:

```
First group of servo  motors
-------------

J: 3
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000    0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000

K: 3
O:      10000  10000      0 -10000  10000
S: 0 0  -10000  -10000      0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000

Second group of servo motors
------------

L: 3
O:      10000  10000      0 -10000  10000
S: 0 1  -10000  -10000      0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000

M: 3
O:      10000  10000      0 -10000  10000
S: 0 1  10000  10000    0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000

Motor speed mixer
-----------------

N: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000
```

but, unexpectedly enough, when run `mixer load /dev/pwm_output0 /etc/mixers/<my mixer file name>` in NSH, the output turns to be

```
O:      10000  10000      0 -10000  10000
S: 0 0  10000  10000    0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000
O:      10000  10000      0 -10000  10000
S: 0 0  -10000  -10000      0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000
O:      10000  10000      0 -10000  10000
S: 0 1  -10000  -10000      0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000
M: 3
O:      10000  10000      0 -10000  10000
S: 0 1  10000  10000    0 -10000  10000
S: 0 2  -10000  -10000      0 -10000  10000
S: 0 3  0       -2000      0 -10000  10000
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000
[px4io] mixer sent
```

it successfully escaped lines begin with J,K,L and N, also all the comments in the mixer file!
how does it works like this?

taogashi, 2015-06-18:
@tumbili forget it, I think this is not the problem. Mixer files are truncate when compiling...

taogashi, 2015-06-18:
Thx a lot

closed 2015-06-21 by taogashi