PX4/PX4-Autopilot #2299: Difficulties with yawrate to lock heading hold

AndreasAntener, 2015-06-07:
@LorenzMeier, @tumbili 
Here's a log from todays flight: http://dash.oznet.ch/view/TcrueUA3R433rTXvWn3vKB
(latest release_v1.0.0 merged with the osd branch)

Yawrate changes a lot during the flight, so it switched always in and out of heading hold (while always in posctl). That caused the heading to converge to 0. Same as I saw in HIL

My yawrate is oscillation a lot, I'm not sure if you can set the threshold high enough for that:
[image]
What do you think?

Here you see the convergence:
[image]

tumbili, 2015-06-07:
@AndreasAntener Yeah I'm stupid, I implemented it the wrong way. I'm fixing it.

tumbili, 2015-06-07:
@AndreasAntener Hope this fixed it https://github.com/PX4/Firmware/pull/2300

AndreasAntener, 2015-06-07:
Cool :) going to try it soon

kd0aij, 2015-06-08:
@AndreasAntener I tested POSCTL on my Ninox flying wing yesterday, and it seemed to work correctly. I had made one small change: #2301 Perhaps there is less tail wag with my plane (how can you wag your tail if you don't have one? call it yaw oscillation :)

[image]

AndreasAntener, 2015-06-08:
;) I know my FX-61 oscillates in yaw quite a bit after turns..

Hm, if I look at your yaw rate, you should have experienced the same problem as me..

kd0aij, 2015-06-08:
you're right; added yawrate to my plot. Same magnitude as yours; +/-0.5 rad/sec
[image]

tumbili, 2015-06-08:
@kd0aij @AndreasAntener Well with the fix I pushed today it shouldn't matter anyway. As long as your yawrate comes close to zero at some point everything is fine. (should be :-))

LorenzMeier, 2015-06-14:
@tumbili Is this one resolved?

tumbili, 2015-06-14:
@LorenzMeier I think so. It worked fine on Thursday when we tested it.

LorenzMeier, 2015-07-05:
Reopening this after some more testing by ada suggests the lock-on still might need some work.

AndreasAntener, 2015-07-06:
http://dash.oznet.ch/view/v98KHkQvYTRnRpEyjrBjPR
Marked the 2 spots here:
[image]
(yaw is scaled by 0.5)

LorenzMeier, 2015-07-06:
I think the fails were actually on other occasions. The location you've marked was counter-steering to a strong motion, but it actually stabilised the course over ground.

[image]

LorenzMeier, 2015-07-06:
I've pushed a fix for the delayed lock here:
https://github.com/PX4/Firmware/commit/fd63ba7b898b95469b62f6f3f42f12e042aae619

Closing this for now.

closed 2015-07-06 by LorenzMeier

AndreasAntener, 2015-07-06:
Ok, I can see the delayed lock because of the bad yaw behaviour of my Phantom after rolls (which I didn't notice in flight).

But about the 2 spots I marked: are you saying this is a tuning issue?
About stabilizing COG: what I'm missing in the log is the heading that it gets locked onto. Then we would see the exact moment of the lock, and of course the COG it wants to hold.