PX4/PX4-Autopilot #2309: PX4Flow POSCTL no longer working in RC11

kd0aij, 2015-06-09:
My F450 quad with PX4Flow is no longer holding position (outdoors with GPS and flow). Flow data is present and quality is high, but position doesn't track the setpoint at all. Altitude hold is very good though in the 1 to 3 meter AGL range. Recalibrating the magnetometer did not improve behavior.

Could this be due to a broken configuration due to recent changes in parameter handling?

Log: http://dash.oznet.ch/view/oLG4ezAyzewqiNzRn8uLWU

tumbili, 2015-06-09:
@kd0aij I can give POSCTRL a try today, without flow although. I assume you are using the release branch?

kd0aij, 2015-06-09:
I used the RC11 tag for this flight. Comparing the log with that from 4 June, I notice that mainstate was 3 before in POSCTL, and in this flight it is 2. Could it be that the flight mode changes in the firmware have broken my configuration? I thought I looked at the flight modes on the bench; definitely verified that I have acro mode now (it works as expected), but maybe I didn't verify that ALTCTL/POSCTL works properly with my TX settings. This would explain my problems as "user error".

kd0aij, 2015-06-09:
Just verified on the bench in QGC that mode switching is working properly; ALTCTL/POSCTL are correctly announced by the TX, as are auto/mission/loiter/rtl. Also manual/acro is working.
BTW, I did some full throttle testing in this flight, and there was never a hint of instability. Also did manual mode descents with no (or very little) wobble.

LorenzMeier, 2015-06-09:
We didn't change anything on the flag handling side, and the fact that GPS continues to work is confusing. It would warrant a code inspection in the multicopter pos controller.

kd0aij, 2015-06-09:
I'll reflash with commit 0c0500 and see if it works again

kd0aij, 2015-06-09:
same behavior with the previously working commit 0c0500. Seems like the problem must be compass calibration or rotation. But hmc5883 test reports external compass, heading looks good in QGC and recalibrating the compass didn't change anything. Not sure where else to look; what else could have caused POSCTL to break completely? It still looks like the position controller is commanding motion in the wrong direction.

LorenzMeier, 2015-06-12:
Any news on this?

LorenzMeier, 2015-07-03:
@kd0aij Any news on this one since?

kd0aij, 2015-07-03:
No, I haven't thought of anything new to try with flow.
But GPS-based POSCTL and auto:waypoints is working properly.

andre-nguyen, 2015-07-18:
If I may give my comments on this issue. I find it more difficult compared to a year ago to have the local_position.valid flag to true using optical flow even when quality is high (~200) over a few seconds. I'm not quite sure how to investigate this. @kd0aij are you seeing the same kind of results?

LorenzMeier, 2015-11-06:
Fixed on master, but still WIP.

closed 2015-11-06 by LorenzMeier