PX4/PX4-Autopilot #2315: set_nav_state() implementation looks fishy

devbharat, 2015-06-10:
In the file 

https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp 

in the implementation of bool set_nav_state(), there is a cascaded switch-case statement with the same expression (status->main_state) as the expression of the outer switch-case statement. Line 538 and 558.

bansiesta, 2015-06-12:
I think the way this works is that in line [538](https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L538) several main states are taken into account and some failsafe action depending on the RC signal is done which happens to be the same for all these states.

If there is no failsafe condition, then the main state is left as is in [558](https://github.com/PX4/Firmware/blob/master/src/modules/commander/state_machine_helper.cpp#L558), so basically copied over using a second switch.

In my opinion this is kind of ugly but not wrong.

devbharat, 2015-06-12:
I don't see the need for the 2nd switch case statement for the code to work as it is working right now(although I don't suppose that is how it is intended to work).

The way it is implemented now, the failsafes are only checked when status->main_state is POSCTL(since if-else is inside the case POSCTL), thus, looking for status->main_state in a switch case statement (line 558) in the else condition makes little sense as it'd always be POSCTL (already filtered by the previous switch case statement). So you might as well jump to line 572 directly.

bansiesta, 2015-06-12:
The status->nav_state still needs to be set, so the second switch is needed.

No, the failsafes are checked for all cases (ACRO, MANUAL, ALTCTL, POSCTL) because the ```cases in the switch fall through if there is no break or return.

devbharat, 2015-06-12:
I agree.

closed 2015-06-12 by devbharat