PX4/PX4-Autopilot #2316: Build POSIX port with LLVM linker

LorenzMeier, 2015-06-10:
Currently the build / param system uses flags which are supported by the GNU linker (GOLD), but not by the LLVM linker. Since "pure" LLVM builds, e.g. on Mac OS, do not support GOLD, we need to find a more standard approach for params to get rid of the section commands LLVM does not understand.

LorenzMeier, 2015-06-10:
@mcharleb This is of medium priority. Its not very relevant as long as we're not trying to roll out to the general development community. Once we do try to, however, its something that we need to resolve. Ultimately it depends on historic architectural decisions so its kind of my problem, but I would appreciate any help / support you can provide on this one.

LorenzMeier, 2015-06-10:
Also please file issues for other architectural limitations that you might have found and assign them to the Linux port milestone.

mcharleb, 2015-06-10:
I started looking at this. I assume you mean MCLinker vs LLD. MCLinker
doesn't support partial linking. Modules could be archives of .o files and
the final link would link the module archives.

LorenzMeier, 2015-06-13:
Great, would be fantastic to look further into this next week. I have fixed the trivial build errors for OS X on the master_m branch. @davids5 Any experience to add?

davids5, 2015-06-13:
@LorenzMeier I am not sure I follow this 100% but, is the only issue the the linker coalescing the parameter segments into one table? 

If so, we could have a different set of macros for the resource rich builds. These would uses a cpp static initializers construct to add the params to a table. Or just use the same mechanisms the tools uses to build it's list of cpp static initializers.

mcharleb, 2015-06-13:
The issue as I understand it is that for each module dir, the build system creates the respective .o files and then prelinks them into module.pre.o and then does an objcopy on them to localize non-exported symbols. This prevents a symbol clash between global variables across modules except for those explicitly exported.

MCLinker doesn't enable you to prelink (no -Ur option). Its not supported. We could just create a module.a from all the .o files but I'm not sure how to replicate the variable scoping behavior.

I never got as far as testing whether MCLinker supported the (partial) linker scripts used to crate the param section.

LorenzMeier, 2015-06-14:
I think we probably want to weight the options in a call - its at least on their agenda now:
https://llvm.org/bugs/show_bug.cgi?id=23024

LorenzMeier, 2015-06-26:
@mcharleb I would be interested to see how bad it is when disabling partial linking for OS X CLANG. I recall doing this and ending up only with a param section linkage fail.

mcharleb, 2015-06-27:
I don't have access to OSX, but we have a DSP toolchain with MCLinker I
could try. I think OSX uses LLD though.

LorenzMeier, 2015-06-27:
I did take an educated guess on Ubuntu, and I get the same error message as on OS X, I guess you can repro with this:

```diff
--- a/makefiles/posix/toolchain_native.mk
+++ b/makefiles/posix/toolchain_native.mk
@@ -92,7 +92,7 @@ DEV_VER_SUPPORTED      = 4.2.1
 endif
 
 #LD                     = ld.gold
-LD                      = ld.bfd
+LD                      = llvm-link-3.5
 AR                      = ar rcs
 NM                      = nm
 OBJCOPY                         = objcopy
```

Which produces this error, similar to the OS X default linker:

```
PRELINK: /home/drone/src/Firmware/Build/posix_sitl.build//home/drone/src/Firmware/src/drivers/blinkm/module.pre.o
llvm-link-3.5: Unknown command line argument '-Ur'.  Try: 'llvm-link-3.5 -help'
llvm-link-3.5: Did you mean '-S'?
```

mcharleb, 2015-06-29:
I gave this some time on the plane ride yesterday. There appears to be no -Ur support for LLD or MCLinker so I know of no way to take a bunch of object files, prelink them into an object file, and then localize all hidden global variables (via objcopy).

We can't build all the files in a module into one .o file directly because there is a mix of C ad C++ files and the C files require C linkage (compiled with clang vs clang++). 

There is no way to compile all the .cpp files in to an object an the .c files into an object and the prelink the two object files together. 

The linker script needs to be run on the combined prelinked modules which again isn't possible.

The requirements to use llvm-link or MCLinker seem to be:

1. Give up on localizing hidden global variables.
2. combine all module xxx.cpp.o and yyy.c.o files into a module.a archive.
3. Fix param to not require use of a linker script

LorenzMeier, 2015-08-17:
@mcharleb We have other things we need to address right now, but medium-term it would be good to go down the route of what you described and enable this.

LorenzMeier, 2015-11-25:
Done!

closed 2015-11-25 by LorenzMeier