PX4/PX4-Autopilot #2319: uORB multi pubsub depends on order of subscribing/publishing

bansiesta, 2015-06-10:
Below is what I found, correct me if I'm just wrong :):

Basically, I think, in the current multi pubsub implementation, the publisher needs to be started before the subscriber, otherwise they won't talk to each other.

An example from #2297 would be:

1. Mavlink starts first and wants to subscribe to the distance_sensor topic. It gets the first instance here:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORB.cpp#L144
2. The the ll40ls wants to do a advertise_multi:
https://github.com/PX4/Firmware/blob/master/src/drivers/ll40ls/LidarLitePWM.cpp#L116
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORB.cpp#L93
I believe this fails here:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBManager_nuttx.cpp#L246
And gets +1 here probably:
https://github.com/PX4/Firmware/blob/master/src/modules/uORB/uORBDevices_nuttx.cpp#L471
3. This leads to mavlink subscribing on 0 and the ll40ls publishing on 1.

Instead of the check ...
- Does the node exist already?

It should be something like:
- Does the node exist already and has stuff been published yet?
- Does the node exist already but is it only opened by a subscriber?

bansiesta, 2015-06-12:
@LorenzMeier bump, is this intended or should I try to come up with a fix?

LorenzMeier, 2015-06-12:
The analysis is correct and we need to fix it, a fix would be more than welcome. I think I just found another race because of this in the HIL setup.

closed 2015-06-16 by bansiesta