PX4/PX4-Autopilot #2323: RTL landing on fixedwing

LorenzMeier, 2015-06-11:
From @kd0aij: RTL flew at 42m above home. On reaching home location announced RTL: landing and continued flying forever at 40% throttle.

Switching off RC resulted in landing sequence.

kd0aij, 2015-06-11:
actually I entered RTL in both cases by switching off the TX
At the first airport in Xplane, the flyaway occurred. After changing airports and restarting QGC/pixhawk, a correct landing sequence occurred. Don't know what was different besides the Xplane location.

LorenzMeier, 2015-06-11:
If I don't switch off the TX RTL works fine via command. So curious to see if it works with TX loss.

LorenzMeier, 2015-06-14:
@kd0aij Would you mind testing in HIL again? We've fixed a bunch of mission related smaller issues which combined could lead to the error you were seeing.

kd0aij, 2015-06-14:
This was working correctly for me in HIL as of 2 days ago :)

LorenzMeier, 2015-06-14:
Ok, closing this then, thanks!

closed 2015-06-14 by LorenzMeier