PX4/PX4-Autopilot #2329: Give user control to define servo midpoint and range

tumbili, 2015-06-11:


LorenzMeier, 2015-06-12:
I think this is the same type of request as someone comes into a car shop and wants new brakes, but what is really the problem is a leak of brake fluid. Could you re-formulate what the actual problem is, and what the users are trying to fix with it?

This here allows to set the min and max values, the trim setup allows to set the midpoint, but much more intuitively than having average Joe doing PWM math.

https://github.com/PX4/Firmware/pull/2340

We should try to focus on what users need, and try to ignore what they had been doing before - its up to us to find the best approach to solve their problem, which is probably setting up the control surfaces on the plane.

tumbili, 2015-06-12:
The user wants to adjust the range of the pwm for each servo separately. The user is used to do this with the remote control (adding range and offset). He couldn't do that anymore because the pwm range was fixed and the mid point was never adjustable. Then he want to use the trim option for fine tuning.

LorenzMeier, 2015-06-12:
Well we don't care what they're used to. Because if we do, we need to replicate all the mistakes and complexity which had been done in the past. We care about the user experience as a whole. I noticed for example that one of the Caipi's had an elevon limitation because of a 30% offset on the servo. This will deteriorate the auto flight performance and has to be fixed on the linkage.
I think we need to get back to the drawing board with a fresh airframe and work out the simplest way to set it up. We then implement whatever is required for that.

kd0aij, 2015-06-12:
I agree that it isn't mechanically optimal to use less than the full available range of each and every servo in a setup. But there are clear cases where this is a very big convenience factor, as in the setup of retracts, bomb drop mechanisms, etc. It may be a *necessity* when matching multiple servos driving a single surface, but I've never done that.
While it isn't *necessary* to do things the same way that experienced RC pilots have done them in the past, it might be the case that those same pilots have more experience than we. 
It seems to me that the existing degrees of freedom in the mixers will serve this purpose, and that we just need to make it practical to adjust those offset and scaling parameters. As it is, adjustment of those parameters is not at all user-friendly.

LorenzMeier, 2015-06-12:
Mark, that makes perfect sense and is certainly the goal. I think we need someone on the GCS side to be happy to take care of bringing the mixer file into an editor window and saving it with one click. Its reasonably easy to do. This PR adds already the ability to query user-settable parameters in the airframe config for the min / max values, which should help together with the trim capability to get average Joe out of trouble.

https://github.com/PX4/Firmware/pull/2340

Note that the airframe config needs to use ```set PWM_MIN p:PWM_MIN``` in order to enable the parameter-driven approach. All airframes default to still using the old system.

tumbili, 2015-06-12:
Basically, if the user can modify the mixer in QGC and if he can set the min/max pwm values then we have everything.

closed 2015-06-15 by tumbili