PX4/PX4-Autopilot #2369: mc_att_control does not publish vehicle_attitude_setpoint in manual mode

kd0aij, 2015-06-14:
but the description of MulticopterAttitudeControl::control_attitude says it does. The only code I find to generate attitude setpoint from manual controls is in mc_pos_control. 

If the comments are not merely out of date, this indicates missing code in mc_att_control. In any case it was a source of confusion for me when modifying the behavior of mc_att_control, since this module currently relies on the position controller to publish manual attitude setpoints.

closed 2015-06-14 by LorenzMeier

kd0aij, 2015-06-14:
that was fast :+1: 
Is there any background info on why pos control is generating manual attitude setpoints?

LorenzMeier, 2015-06-14:
It should be in a neutral stick mapper component, but it doesn't make sense to have it in the controller. The position controller is at least one level up, which is going into the right direction.

kd0aij, 2015-06-14:
That actually makes sense. I think I'm beginning to understand the architecture well enough to ask the right questions. I attempted a hack to att control to override acro mode based on roll/pitch stick inputs, but found I had to put it into commander instead because of this (or duplicate att sp code).

commit 9e3e43c4 (2015-06-14): Update comments in attitude controller. Fixes #2369