PX4/PX4-Autopilot #2379: IRLock: use LANDING_TARGET for data publish?

TSC21, 2015-06-15:
IRLock driver is implemented but not being used anywhere. If one wants to use it onboard, and possibly analyse data offboard, does `LANDING_TARGET` mavlink msg fit the job? (assuming that uORB topic will be created with similar data fields). Checking out the fields, I can see on driver:
```c++
uint64_t timestamp; /** microseconds since system start **/
uint16_t target_num;	/** target number prioritised by size (largest is 0) **/
float angle_x;	/** x-axis angle in radians from center of image to center of target **/
float angle_y;	/** y-axis angle in radians from center of image to center of target **/
float size_x;	/** size in radians of target along x-axis **/
float size_y;	/** size in radians of target along y-axis **/
```
in the case of `LANDING_TARGET`:
```c++
uint8_t target_num /** The ID of the target if multiple targets are present **/
uint8_t frame /** MAV_FRAME enum specifying the whether the following fields are earth-frame, body-frame, etc. **/
float angle_x /** X-axis angular offset (in radians) of the target from the center of the image **/
float angle_y /** Y-axis angular offset (in radians) of the target from the center of the image **/
float distance /** Distance to the target from the vehicle in meters **/
So if we want to complement both data msg data fields, how can we set this up? I suppose IRLock only works with one target, so ID will always be the same. `frame` can be easily fill. Both `angles` are correspondent. But then we have `distance`: assuming my line of thought is correct, this can be calculated with trigonometry using both X and Y angles and distance to ground.
What's your thoughts @LorenzMeier, @rmackay9?

TSC21, 2015-06-15:
Better thought: shouldn't `distance` be a XYZ array (or LLA) so it can be fed into `LANDING` state of navigation, for example?

LorenzMeier, 2015-06-16:
The transformation from angle to position requires the camera to vehicle body frame transformation. That's why its presented as raw data.

rmackay9, 2015-06-16:
Yes, the way it's implemented the camera only has to present the very basic data that it can easily get (size and location of the object from it's point of view).  This makes it very easy to implement from the camera's point of view and for the autopilot, well, it's already got to do all that work for the IR-Lock sensor anyway so might as well also use it for the cases where it's using a camera instead of IR-Lock.

TSC21, 2015-06-16:
@rmackay9 thanks for the feedback! I also think that https://github.com/mavlink/mavros/pull/311/files#r32582378 is relevant to this matter and should be a thing to improve.

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier