PX4/PX4-Autopilot #2381: UAVCAN broken in stable / master

mhkabir, 2015-06-15:
The UAVCAN dynamic node ID allocation changes got into stable, which increases memory consumption to a breaking point, and also renders the system unusable.

This patch gets it back online and usable 
```
diff --git a/src/modules/uavcan/uavcan_main.cpp b/src/modules/uavcan/uavcan_main.cpp
index 2f3e1d5..85f55f9 100644
--- a/src/modules/uavcan/uavcan_main.cpp
+++ b/src/modules/uavcan/uavcan_main.cpp
@@ -53,10 +53,6 @@
 #include <drivers/drv_pwm_output.h>
 
 #include "uavcan_main.hpp"
-#include <uavcan_posix/dynamic_node_id_server/file_event_tracer.hpp>
-#include <uavcan_posix/dynamic_node_id_server/file_storage_backend.hpp>
-
-#include <uavcan_posix/firmware_version_checker.hpp>
 
 //todo:The Inclusion of file_server_backend is killing
 // #include <sys/types.h> and leaving OK undefined
@@ -75,20 +71,12 @@
  * UavcanNode
  */
 UavcanNode *UavcanNode::_instance;
-uavcan::dynamic_node_id_server::CentralizedServer *UavcanNode::_server_instance;
-uavcan_posix::dynamic_node_id_server::FileEventTracer tracer;
-uavcan_posix::dynamic_node_id_server::FileStorageBackend storage_backend;
-uavcan_posix::FirmwareVersionChecker fw_version_checker;
 
 UavcanNode::UavcanNode(uavcan::ICanDriver &can_driver, uavcan::ISystemClock &system_clock) :
 	CDev("uavcan", UAVCAN_DEVICE_PATH),
 	_node(can_driver, system_clock),
 	_node_mutex(),
-	_esc_controller(_node),
-	_fileserver_backend(_node),
-	_node_info_retriever(_node),
-	_fw_upgrade_trigger(_node, fw_version_checker),
-	_fw_server(_node, _fileserver_backend)
+	_esc_controller(_node)
 {
 	_control_topics[0] = ORB_ID(actuator_controls_0);
 	_control_topics[1] = ORB_ID(actuator_controls_1);
@@ -154,7 +142,6 @@ UavcanNode::~UavcanNode()
 	perf_free(_perfcnt_node_spin_elapsed);
 	perf_free(_perfcnt_esc_mixer_output_elapsed);
 	perf_free(_perfcnt_esc_mixer_total_elapsed);
-	delete(_server_instance);
 
 }
 
@@ -305,74 +292,6 @@ int UavcanNode::init(uavcan::NodeID node_id)
 		br = br->getSibling();
 	}
 
-
-	/* Initialize the fw version checker.
-	* giving it it's path
-	*/
-
-	ret = fw_version_checker.createFwPaths(UAVCAN_FIRMWARE_PATH);
-
-	if (ret < 0) {
-		return ret;
-	}
-
-
-	/* Start fw file server back */
-
-	ret = _fw_server.start();
-
-	if (ret < 0) {
-		return ret;
-	}
-
-	/* Initialize storage back end for the node allocator using UAVCAN_NODE_DB_PATH directory */
-
-	ret = storage_backend.init(UAVCAN_NODE_DB_PATH);
-
-	if (ret < 0) {
-		return ret;
-	}
-
-	/* Initialize trace in the UAVCAN_NODE_DB_PATH directory */
-
-	ret = tracer.init(UAVCAN_LOG_FILE);
-
-	if (ret < 0) {
-		return ret;
-	}
-
-	/* Create dynamic node id server for the Firmware updates directory */
-
-	_server_instance = new uavcan::dynamic_node_id_server::CentralizedServer(_node, storage_backend, tracer);
-
-	if (_server_instance == 0) {
-		return -ENOMEM;
-	}
-
-	/* Initialize the dynamic node id server  */
-	ret = _server_instance->init(_node.getNodeStatusProvider().getHardwareVersion().unique_id);
-
-	if (ret < 0) {
-		return ret;
-	}
-
-	/* Start node info retriever to fetch node info from new nodes */
-
-	ret = _node_info_retriever.start();
-
-	if (ret < 0) {
-		return ret;
-	}
-
-
-	/* Start the fw version checker   */
-
-	ret = _fw_upgrade_trigger.start(_node_info_retriever, fw_version_checker.getFirmwarePath());
-
-	if (ret < 0) {
-		return ret;
-	}
-
 	/*  Start the Node   */
 
 	return _node.start();
diff --git a/src/modules/uavcan/uavcan_main.hpp b/src/modules/uavcan/uavcan_main.hpp
index dc81fb5..1157843 100644
--- a/src/modules/uavcan/uavcan_main.hpp
+++ b/src/modules/uavcan/uavcan_main.hpp
@@ -48,13 +48,6 @@
 #include "sensors/sensor_bridge.hpp"
 
 
-#include <uavcan/protocol/dynamic_node_id_server/centralized.hpp>
-#include <uavcan/protocol/node_info_retriever.hpp>
-#include <uavcan_posix/basic_file_server_backend.hpp>
-#include <uavcan/protocol/firmware_update_trigger.hpp>
-#include <uavcan/protocol/file_server.hpp>
-
-
 /**
  * @file uavcan_main.hpp
  *
@@ -83,7 +76,7 @@ class UavcanNode : public device::CDev
 
 	static constexpr unsigned PollTimeoutMs      = 10;
 
-	static constexpr unsigned MemPoolSize        = 10752; ///< Refer to the libuavcan manual to learn why
+	static constexpr unsigned MemPoolSize        = 7168; ///< Refer to the libuavcan manual to learn why
 
 	/*
 	 * This memory is reserved for uavcan to use for queuing CAN frames.
@@ -147,7 +140,6 @@ private:
 	unsigned		_output_count = 0;		///< number of actuators currently available
 
 	static UavcanNode	*_instance;			///< singleton pointer
-	static uavcan::dynamic_node_id_server::CentralizedServer *_server_instance;              ///< server singleton pointer
 
 	Node			_node;				///< library instance
 	pthread_mutex_t		_node_mutex;
@@ -155,11 +147,6 @@ private:
 	UavcanEscController	_esc_controller;
 
 
-	uavcan_posix::BasicFileSeverBackend _fileserver_backend;
-	uavcan::NodeInfoRetriever  _node_info_retriever;
-	uavcan::FirmwareUpdateTrigger  _fw_upgrade_trigger;
-	uavcan::BasicFileServer        _fw_server;
-
 	List<IUavcanSensorBridge *> _sensor_bridges;		///< List of active sensor bridges
 
 	MixerGroup		*_mixers = nullptr;
```

LorenzMeier, 2015-06-15:
@pavel-kirienko We will relatively soon switch from an agile development model in terms of user releases to the currently established more stable baseline. It would be great if we could look tomorrow into the RAM usage and stability issues here, as the merge window is closing. The current memory consumption of UAVCAN doesn't allow it to launch it as default module and we already need to de-feature the system (reduce e.g. buffer sizes for the logging system) to fit it when its running.

pavel-kirienko, 2015-06-15:
@LorenzMeier I have already proposed a solution here https://github.com/UAVCAN/libuavcan/issues/33: make the core UAVCAN functionality enabled by default, whereas the dynamic node ID allocator and firmware upgrade master would be optional. I'm well aware of usability penalties, but so far this is the only solution that is radical enough.

Also, when my suggestions described [here](https://github.com/UAVCAN/libuavcan/issues/33#issuecomment-112053842) are implemented in the `uavcan` driver, the memory use will further drop by 2~3K (although that's not enough to be a solution).

> look tomorrow into the RAM usage and stability issues here

Are you proposing to have a call?

@mhkabir Is it the original patch I sent you two days ago?

davids5, 2015-06-15:
@LorenzMeier , @pavel-kirienko 

What time UTC should we have a call to define the short term and long term goals that will best align with the new release strategy?

davids5, 2015-06-17:
PR https://github.com/PX4/Firmware/pull/2404 is pushsed for short term conditional removal. Working on long term solution next

davids5, 2015-06-17:
@pavel-kirienko , @LorenzMeier 

Results of UAVCAN_MAX_NETWORK_SIZE_HINT testing

No change in elf sections sizes, but reducing  the allocation from 64 to UAVCAN_MAX_NETWORK_SIZE_HINT=16 saved **5248** Bytes!

[image]

LorenzMeier, 2015-06-17:
Ok, can we get that as PR on top of the current changes?

LorenzMeier, 2015-06-17:
@pavel-kirienko this is what the MAVLink app is consuming - it routes all information of the system at 12K usage. In my view already pretty heavy, but it does missions / FTP / all system streams. I did already remove unnecessary stack sizes and buffers.

```
nsh> free
             total       used       free    largest
Mem:        225968     174352      51616      50352
nsh> mavlink start -d /dev/ttyS2
mavlink_if1: mode: 0, data rate: 2880 B/s on /dev/ttyS2 @ 57600B
nsh> mavlink_if1: Disabling hardware flow control
free
             total       used       free    largest
Mem:        225968     186080      39888      36256
```

Now when I start uavcan, I'm out of RAM:

```
free
             total       used       free    largest
Mem:        225968     186080      39888      36256
nsh> 
nsh> 
nsh> uavcan start
uavcan: Node ID 1, bitrate 1000000
uavcan: SW version vcs_commit: 0x959333d6
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        225968     210304      15664      10624
```

Note that the sdlog2 app will now fail to log, as it doesn't have enough RAM for its log buffer. So we really need to get UAVCAN closer to the 12-14K mark to fit.

davids5, 2015-06-17:
@pavel-kirienko does 

static constexpr unsigned MemPoolSize        = 10752; ///< Refer to the libuavcan manual to learn why

Still make sence after all the changes you did?

pavel-kirienko, 2015-06-18:
@davids5 I introduced some memory optimizations today, particualrly:

1. Removed the allocation table cache from the centralized server implementation, which saved 2144 bytes. The distributed server is still disgustingly huge though.
2. Fixed structure packing - it used to break things and increase code size because it was used inappropriately (so it could not be enabled), now it's organized properly (and it should be safe to enable  it).
3. Rearranged some class fields to reduce padding. This change together with the previous one allowed to reduce block size to 48 bytes on ARM, which is a huge win (didn't test on any real hardware yet though).

This is what I suggest to add to the module defines:

```
-DUAVCAN_MEM_POOL_BLOCK_SIZE=48
-DUAVCAN_PACKED_BEGIN='_Pragma("pack(1)")' 
-DUAVCAN_PACKED_END='_Pragma("pack()")'
```

The pool size can be reduced for now:

```c++
static constexpr unsigned MemPoolSize = 64 * uavcan::MemPoolBlockSize;
```

Hopefully it should work.

@LorenzMeier @davids5 May I suggest to use `-fpacked-enums` to reduce general memory footprint? (not only libuavcan)

LorenzMeier, 2015-06-18:
That sounds like good progress! Inline with what we discussed yesterday it would be great if you could also test this on the actual hardware. The main motivation is because it will help the protocol development to fit the embedded space.

LorenzMeier, 2015-06-18:
@pavel-kirienko I will look into ```-fpacked-enums```, but changes like this need to be vetted very carefully when done for the full system. Also because we probably need to apply it to the OS.

pavel-kirienko, 2015-06-18:
@LorenzMeier Yep, I will set up a testing rig here. The packing turned out to hard fault the system so 48 byte blocks are still not acceptable, but I'll be working on it some more tomorrow and hopefully a way will be devised to get the memory consumption down to the 14K limit you set above.

pavel-kirienko, 2015-06-20:
OK, I succeeded in reducing the pool block size to 48 bytes through some optimizations deep inside libuavcan. Current head of `uavcan_size_investigation`:

```
nsh> free
             total       used       free    largest
Mem:        223888      29360     194528     154944
nsh> uavcan start
<...>
nsh> free
             total       used       free    largest
Mem:        223888      44160     179728     154944
```

14800 bytes.

LorenzMeier, 2015-06-20:
Awesome! @pavel-kirienko What is the testing state of this - is this ready for the stable branch?

pavel-kirienko, 2015-06-20:
libuavcan is, but I'm not sure about the rest of the branch - let's hear from @davids5 who created it initially.

pavel-kirienko, 2015-06-21:
UAVCAN firmware update logic takes 27k.

```
nsh> free
             total       used       free    largest
Mem:        224336      28256     196080     156608
nsh> uavcan start
<...>
nsh> free
             total       used       free    largest
Mem:        224336      41760     182576     156608
nsh> uavcan start fw
nsh> free
             total       used       free    largest
Mem:        224336      69152     155184     133184
```

@davids5 Hanging we discussed last night is has been fixed via  https://github.com/PX4/Firmware/commit/42eb1d17c08657e5af7d08f7a1f347af064ea9ce.

pavel-kirienko, 2015-06-21:
After some adjustments, the firmware update logic takes 26K (see `nuttx_next_uavcan_multi_threaded`).

davids5, 2015-06-21:
Here are the results for nuttx_next_uavcan_multi_threaded 853b9dc567abcefcf7ee111492a5fff27fad53ce 

[image]

davids5, 2015-06-21:
Here are the figures for the px4cannode

[image]

pavel-kirienko, 2015-06-21:
I wonder why do our results differ by 1k.

davids5, 2015-06-21:
@pavel-kirienko 

Could it be the 128 vs 64 magic number in

```static constexpr unsigned MemPoolSize = 64 * uavcan::MemPoolBlockSize;```

Or the fact I get a CCASERT failure with 48 and have to use 56?

davids5, 2015-06-21:
Here are the results on master_uavcan_modular	500a34749518b8853d2ba2849515d76f666bb5cd	

You will note an oddity  - what looks like a 592 byte mem leak after a stop, but a restart and stop of the fw thread returns to the original. - I will recheck the lifecycle management to see if I missed anything.

[image]

The main uavcan task when stopped look to leak 928 bytes. - think this may have been in the original master, and I will look into it also.

[image]

LorenzMeier, 2015-06-21:
Ok. When you both think its ready for prime-time it would be great to have a PR against master. Because of the shallow testing I do not want to launch it by default, however, it should launch for all configs when the UAVCAN_ENABLED param is set to 1. We will later then default it to on. This will allow uavcan users to enable it from the GUI, but still shield the release from any side-effects we might have been missing.

pavel-kirienko, 2015-06-21:
Okay. I shudder to ask, how much time do we have until the release?

LorenzMeier, 2015-06-21:
There are not many issues left to resolve. We currently shoot for the beginning of next week. https://github.com/PX4/Firmware/milestones/First%20beta%20release

pavel-kirienko, 2015-06-21:
Oh no. You said that the next release will follow in a few weeks, does it still hold?

LorenzMeier, 2015-06-21:
Well, do you want to have the first broadly available release be conceived as non-operational with UAVCAN? Its not so much a time question but "first impression" question. To me it sounds like a terrible first impression to release with the current state.

Also I can't make any assertions when the next release will happen, and when affected users will upgrade.

davids5, 2015-06-21:
@pavel-kirienko 

As we have been skype chatting about, the memory reduction of using 48 block size will not compile on master_uavcan_modular nor nuttx_next_uavcan_multi_threaded. I think the block allocation rework did not take into account the code you added for the virtual can driver and the testing was only done on uavcan_size_investigation that does not have the virtual can driver.

```In instantiation of 'static void uavcan::IsDynamicallyAllocatable<T>::check() [with T = Queue<VirtualCanIface::RxItem>::Item]```

The issues it the size of Queue<RxItem> rx_queue_ is causing a check failure (CCASERT) on     ```char dummy[(sizeof(T) <= MemPoolBlockSize) ? 1 : -1] = { '0' };```

Please fix the issue and ensure both of the above mentioned branches compile with the uavcan

pavel-kirienko, 2015-06-22:
Fixed in https://github.com/PX4/Firmware/commit/cfde1b74e3df14d16adbbcaf6c9c502363a58e57.

```
nsh> free
             total       used       free    largest
Mem:        224336      28256     196080     156608
nsh> uavcan start
uavcan: Node ID 1, bitrate 1000000
uavcan: SW version vcs_commit: 0x853b9dc5
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        224336      41248     183088     156608
nsh> uavcan start fw
nsh> free
             total       used       free    largest
Mem:        224336      62512     161824     156608
```

Core: 12992 bytes
Servers: 21264 bytes

davids5, 2015-06-22:
@pavel-kirienko I pulled in the files from https://github.com/PX4/Firmware/commit/cfde1b74e3df14d16adbbcaf6c9c502363a58e57

updated to latest uavcan master

Tested again on master_uavcan_modular

x + SYS_AUTOCONFIG [298,523] : 0
x + SYS_AUTOSTART [299,524] : 4010

```
nsh> free
             total       used       free    largest
Mem:        222272     167184      55088      53024
nsh>
nsh>
nsh>
nsh>
nsh>
nsh> uavcan start
uavcan: Node ID 1, bitrate 1000000
uavcan: SW version vcs_commit: 0xe967ee09
uavcan: sensor bridge 'gnss' init ok
uavcan: sensor bridge 'mag' init ok
uavcan: sensor bridge 'baro' init ok
nsh> free
             total       used       free    largest
Mem:        222272     182608      39664      34656
nsh> uavcan start fw
nsh> free
             total       used       free    largest
Mem:        222272     203856      18416      10640
nsh> uavcan stop fw
nsh> free
             total       used       free    largest
Mem:        222272     182096      40176      34656
nsh> uavcan stop
uavcan: exiting.
nsh> free
             total       used       free    largest
Mem:        222272     169296      52976      34720

```
[image]

I get Core of 15424 and server of 21248 for a total of 36672 bytes

Would you please test master_uavcan_modular, build and configuration and see if you get the same numbers?

davids5, 2015-06-22:
@pavel-kirienko 

The Latest code is not sending

[image]

davids5, 2015-06-23:
@pavel-kirienko 

The first breaking point is in the node ID allocation. The node is stuck not node allocating.

The virtual can driver's send is never being called.

- If I back out that last 2 (see below) changes on nuttx_next_uavcan_multi_threaded  by checking out at 853b9dc567abcefcf7ee111492a5fff27fad53ce and leave the uavcan on master but add the uint8 to nodesatus, etal  (6002a79ef04210ba937a525b42db0bfebc3b128c)  I can boot load. 
See nuttx_next_uavcan_multi_threaded_last_working_protocol_20150622 it is as I described above and works.

However nuttx_next_uavcan_multi_threaded_protocol_20150622 which has the protocol changes AND the virtual can changes will not get a node allocation. 
```
commit f8eefc13f6cbccc8a66901eb1bed7f74d1bd0844
Author: Pavel Kirienko <pavel.kirienko@gmail.com>
Date:   Mon Jun 22 20:17:38 2015 +0300

    uavcannode: makefile update: 48 byte pool blocks, pragmas removed

commit cfde1b74e3df14d16adbbcaf6c9c502363a58e57
Author: Pavel Kirienko <pavel.kirienko@gmail.com>
Date:   Mon Jun 22 20:12:22 2015 +0300

    UAVCAN: 48 byte pool block size
```

I would suggest you fix nuttx_next_uavcan_multi_threaded then cherry pick 6002a79ef04210ba937a525b42db0bfebc3b128c on to it.

Both  nuttx_next_uavcan_multi_threaded_last_working_protocol_20150622 and nuttx_next_uavcan_multi_threaded_protocol_20150622 are throw away.

LorenzMeier, 2015-06-23:
@pavel-kirienko I would appreciate if we could move to a workflow where each and every protocol or library change is put through a minimal bench test suite (boot loading a node and checking if the baro / gps / mag still work) and larger changes are test flown.
Starting next week I might be able to help automate this.

This is not so much a release thing, but due diligence in avionics software development, and avoiding to frustrate application level developers and users of UAVCAN with perceived instability. We spend a significant effort in PX4 on this, and its a cornerstone of the platform adoption - I believe the same will apply for UAVCAN. Your ability to retain control of the library development will depend on it, as your adopters will just refuse to update the library forever if things are breaking regularly.

davids5, 2015-06-23:
Merged and cross checked - @pavel-kirienko changes - Thank you!

Final results
[image]

There are 2 more changes that need to come in:

- [ ] Adding command line option upgrade fw - The reason for this is the bootloader timeout is 5 Seconds. With a fmu on usb the nodes will be booted before the uavcan / fw server is running. This command will cause all devices to report the their current fw CRC and if we have a fw file for that device on the SD card and it has a different CRC than the node, the node will be told to begin_firmware_update
- [ ] **BREAKING CHANGE* Frame format changes

LorenzMeier, 2015-06-25:
Any reason to not check all nodes and upgrade them without a command line upgrade call? Has the boot loader been built to support power-cycling during the flashing (a.k.a write zero to the first atomic memory block initially and write the first block last with the correct address)?

I think its fine for the release to require the command line, but if its a quick thing it might make sense to just automate it right away and make sure the boot loader survives real usage scenarios.

davids5, 2015-06-25:
Yes the bootloader will support  power-cycling during the flashing. They should auto update at boot. 

I am supporting development with ```uavcan update fw``` and future support when mavlink ftp is used to upload new FW to the SD card - that can uses ```uavcan update fw``` to not require a power cycle.

Bootloader works like this:

If there is no valid Application program the nodes will wait. Nodes that auto baud will wait for a message on the can bus.  (It may come from a node that did not auto baud.)

Then the nodes wait for node ID allocation up to 5 seconds. If there is a time out the the node will boot it application.

Given the boot times. The nodes could be in their applications before the start up script finishes.

I am just adding the code to reboot to the BL in the cannode application now and testing it.

This is the start up
```
uavcan start
uavcan start fw
```

```uavcan start``` <- can auto baud
```uavcan start fw``` <- can Node ID allocate - nodes should update FW here or when plugged in or reset after this step in the boot scripts.

davids5, 2015-06-25:
If we need to perform ```uavcan start fw``` and ```uavcan stop fw``` on events Then I just need to understand what the events are. Breaking them out was to ensure memory was not exhausted.

LorenzMeier, 2015-06-25:
We want to scan the bus, check that all nodes have the right firmware version and upgrade all which haven't. The firmware binary can come in as file from microSD or as byte stream via telemetry.

pavel-kirienko, 2015-06-25:
> The firmware binary can come in as file from microSD or as byte stream via telemetry.

The byte stream will have to be stored somewhere first, which reduces the problem to the general case of uploading a file from SD card.

davids5, 2015-06-25:
I have completed and tested the download on new fw booting and running with both FMU and NODE leading and lagging. All cases work and the 

[image]

```
 text    data     bss     dec     hex filename
   7899       0    3552   11451    2cbb Build/px4cannode-v1_bootloader.build/firmware.elf
   text    data     bss     dec     hex filename
 102736     416    3772  106924   1a1ac Build/px4cannode-v1_default.build/firmware.elf
   text    data     bss     dec     hex filename
 910456    2696   34620  947772   e763c Build/px4fmu-v2_default.build/firmware.elf
   text    data     bss     dec     hex filename
  47229     516    3040   50785    c661 Build/px4io-v2_default.build/firmware.elf
```

Some test cases
```
nsh>   cd /fs/microsd/fw
nsh>  rm c/6ac0fd32.bin
nsh: rm: unlink failed: No such file or directory
nsh>  cp 1.0.6ac0fd32.uavcan.bin org.pixhawk.px4cannode-v1/1.0/6ac0fd32.bin
nsh>  cp 1.0.6ac0fd32.uavcan.bin c/6ac0fd32.bin
nsh> uavcan update fw
nsh>
nsh>  rm c/6ac0fd32.bin
nsh>  cp 0.1.6ac0fd32.uavcan.bin org.pixhawk.px4cannode-v1/1.0/6ac0fd32.bin
nsh>  cp 0.1.6ac0fd32.uavcan.bin c/6ac0fd32.bin
nsh>  uavcan update fw
```

Even hit reset in the middle of the download. Recovered just fine :)

LorenzMeier, 2015-06-25:
Awesome! So I'm looking forward to the last few UAVCAN updates and I'll start to worry in parallel about how to get the binaries from the internet there.

pavel-kirienko, 2015-06-25:
I think the specification and the PX4 project in particular could benefit if someone drafted a proposition of how to retrieve firmware files from vendor's websites. Remember that UAVCAN node names are reversed internet domain names, e.g. `org.pixhawk.foobar`, which could be exploited neatly. I don't expect this to become a part of UAVCAN specification until maybe much later, when the idea is more tested and validated against the real world. We actually discussed it a while ago with @davids5, but no follow up happened since then.

LorenzMeier, 2015-06-25:
Agreed. I'm working on that already.

LorenzMeier, 2015-08-14:
Fixed in beta - finally. Needs still an update for the GPS module.

closed 2015-08-14 by LorenzMeier