PX4/PX4-Autopilot #2384: POSIX: Boot fails

LorenzMeier, 2015-06-15:
@mcharleb This is the current status on master when building the posix target and running ```make sitlrun```.

```
Tools/sitl_run.sh
INFO  Shell id is 47843067809472
App name: mainapp
----------------------------------
Running: uorb
----------------------------------
Running: param
----------------------------------
Running: mavlink
arg 0 mavlink
arg 1 start
arg 2 -u
arg 3 14556
WARN  mode: 0, data rate: 2880 B/s on udp port 14556 (file /home/drone/src/Firmware/src/modules/mavlink/mavlink_main.cpp line 1480)
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED
telem> PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
telem> PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
telem> PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
INFO  Setting up UDP w/port 14556

WARN  bind failed
 (file /home/drone/src/Firmware/src/modules/mavlink/mavlink_main.cpp line 963)
WARN  offboard mission init: ERROR (file /home/drone/src/Firmware/src/modules/mavlink/mavlink_mission.cpp line 126)
----------------------------------
Running: simulator
arg 0 simulator
arg 1 start
arg 2 -s
----------------------------------
Running: param
+ CAL_GYRO0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC1_ID: unchanged
----------------------------------
Running: param
+ CAL_MAG0_ID: unchanged
----------------------------------
Running: rgbled
----------------------------------
Running: tone_alarm
INFO  TONE_SET_ALARM 0
----------------------------------
Running: gyrosim
----------------------------------
Running: accelsim
----------------------------------
Running: barosim
INFO  Simulator started
WARN  bind failed
 (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 325)
----------------------------------
Running: adcsim
----------------------------------
Running: commander
INFO  IOCTL DEVIOCGDEVICEID 196608
----------------------------------
Running: sensors
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  TONE_SET_ALARM 9
INFO  ToneAlarm::start_tune
WARN  INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY (file /home/drone/src/Firmware/src/modules/commander/state_machine_helper.cpp line 274)
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
----------------------------------
Running: ekf_att_pos_estimator
.
----------------------------------
Running: mc_pos_control
----------------------------------
Running: mc_att_control
----------------------------------
Running: hil
INFO  <hilservo> 
INFO  MODE_4PWM
INFO  [hilservo] 
INFO  default PWM output device
----------------------------------
Running: dataman
WARN  Incompatible data manager file rootfs/fs/microsd/dataman, resetting it (file /home/drone/src/Firmware/src/modules/dataman/dataman.c line 666)
INFO  [hilservo] 
INFO  starting
dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 101504 bytes
----------------------------------
Running: navigator
----------------------------------
Running: param
+ MAV_TYPE: unchanged
----------------------------------
Running: param
+ RC1_MAX: unchanged
----------------------------------
Running: param
+ RC1_MIN: unchanged
----------------------------------
Running: param
+ RC1_TRIM: unchanged
----------------------------------
Running: param
+ RC1_REV: unchanged
----------------------------------
Running: param
+ RC2_MAX: unchanged
----------------------------------
Running: param
+ RC2_MIN: unchanged
----------------------------------
Running: param
  RC2_TRIM: unchanged
----------------------------------
Running: param
+ RC3_MAX: unchanged
----------------------------------
Running: param
+ RC3_MIN: unchanged
----------------------------------
Running: param
+ RC3_TRIM: unchanged
----------------------------------
Running: param
+ RC4_MAX: unchanged
----------------------------------
Running: param
+ RC4_MIN: unchanged
----------------------------------
Running: param
+ RC4_TRIM: unchanged
----------------------------------
Running: param
+ RC4_REV: unchanged
----------------------------------
Running: param
+ RC6_MAX: unchanged
----------------------------------
Running: param
+ RC6_MIN: unchanged
----------------------------------
Running: param
+ RC6_TRIM: unchanged
----------------------------------
Running: param
+ RC_CHAN_CNT: unchanged
----------------------------------
Running: param
+ RC_MAP_MODE_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_POSCTL_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_RETURN_SW: unchanged
----------------------------------
Running: param
+ MC_PITCHRATE_P: unchanged
----------------------------------
Running: param
+ MC_ROLLRATE_P: unchanged
----------------------------------
Running: mixer
WARN  offboard mission update failed (file /home/drone/src/Firmware/src/modules/navigator/mission.cpp line 238)
WARN  ERROR: invalid mission state (file /home/drone/src/Firmware/src/modules/navigator/mission.cpp line 771)
WARN  node_open as advertiser failed. (file /home/drone/src/Firmware/src/modules/uORB/uORBManager_posix.cpp line 98)
WARN  node_open as advertiser failed. (file /home/drone/src/Firmware/src/modules/uORB/uORBManager_posix.cpp line 98)
WARN  can't open /dev/pwm_output0
 (file /home/drone/src/Firmware/src/systemcmds/mixer/mixer.cpp line 111)
WARN  failed to load mixer (file /home/drone/src/Firmware/src/systemcmds/mixer/mixer.cpp line 80)
Enter a command and its args:
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 10134
```

LorenzMeier, 2015-06-15:
@tumbili Can you reproduce this?

tumbili, 2015-06-15:
No

tumbili, 2015-06-15:
I don't get all the  bind errors

tumbili, 2015-06-15:
Let me just try several times and see if there are issues. On the first attempt the simulation started without any problems.

tumbili, 2015-06-15:
@LorenzMeier @mcharleb Tested master, the simulation runs fine (tested 10 times in a row). But as soon as I have QGC already connected to the port without the simulation having being started then I get these messages again: WARN  node_open as advertiser failed. (file /home/roman/Firmware/src/modules/uORB/uORBManager_posix.cpp line 98)

tumbili, 2015-06-15:
I should also mention that I'm using an older version of jMAVSim 8aaf29c12af0cd55c58d3
I'll try to update and see what that does.

LorenzMeier, 2015-06-15:
Could you try without QGC and jMAVSim running?

tumbili, 2015-06-15:
Works fine for me

LorenzMeier, 2015-06-15:
odd, ok

tumbili, 2015-06-15:
'''
roman@roman-HP-EliteBook-8560p:~/Firmware$ make sitlrun
Tools/sitl_run.sh
INFO  Shell id is 46986347925184
App name: mainapp
----------------------------------
Running: uorb
----------------------------------
Running: param
----------------------------------
Running: mavlink
arg 0 mavlink
arg 1 start
arg 2 -u
arg 3 14556
WARN  mode: 0, data rate: 2880 B/s on udp port 14556 (file /home/roman/Firmware/src/modules/mavlink/mavlink_main.cpp line 1480)
INFO  Setting up UDP w/port 14556

----------------------------------
Running: simulator
arg 0 simulator
arg 1 start
arg 2 -s
----------------------------------
Running: param
+ CAL_GYRO0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC1_ID: unchanged
----------------------------------
Running: param
+ CAL_MAG0_ID: unchanged
----------------------------------
Running: rgbled
INFO  Simulator started
----------------------------------
Running: tone_alarm
INFO  TONE_SET_ALARM 0
----------------------------------
Running: gyrosim
WARN  failed to open /dev/ttyACM0 (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 348)
WARN  failed to send streaming command to /dev/ttyACM0 (file /home/roman/Firmware/src/modules/simulator/simulator_mavlink.cpp line 356)
----------------------------------
Running: accelsim
----------------------------------
Running: barosim
----------------------------------
Running: adcsim
----------------------------------
Running: gps_sim
INFO  <gps> 
INFO  lock
INFO  <gps> 
INFO  unlock
----------------------------------
Running: commander
INFO  IOCTL DEVIOCGDEVICEID 196608
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED
INFO  IOCTL DEVIOCGDEVICEID 1376256
telem> PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
----------------------------------
Running: sensors
INFO  IOCTL DEVIOCGDEVICEID 1310720
telem> PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
INFO  IOCTL DEVIOCGDEVICEID 2293760
telem> PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  TONE_SET_ALARM 9
INFO  ToneAlarm::start_tune
WARN  INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY (file /home/roman/Firmware/src/modules/commander/state_machine_helper.cpp line 274)
----------------------------------
Running: ekf_att_pos_estimator
.
----------------------------------
Running: mc_pos_control
----------------------------------
Running: mc_att_control
----------------------------------
Running: hil
INFO  <hilservo> 
INFO  MODE_4PWM
INFO  [hilservo] 
INFO  default PWM output device
----------------------------------
Running: dataman
INFO  [hilservo] 
INFO  starting
WARN  Incompatible data manager file rootfs/fs/microsd/dataman, resetting it (file /home/roman/Firmware/src/modules/dataman/dataman.c line 666)
dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 101504 bytes
----------------------------------
Running: navigator
----------------------------------
Running: param
+ MAV_TYPE: unchanged
----------------------------------
Running: param
+ RC1_MAX: unchanged
----------------------------------
Running: param
+ RC1_MIN: unchanged
----------------------------------
Running: param
+ RC1_TRIM: unchanged
----------------------------------
Running: param
+ RC1_REV: unchanged
----------------------------------
Running: param
+ RC2_MAX: unchanged
----------------------------------
Running: param
+ RC2_MIN: unchanged
----------------------------------
Running: param
  RC2_TRIM: unchanged
----------------------------------
Running: param
+ RC3_MAX: unchanged
----------------------------------
Running: param
+ RC3_MIN: unchanged
----------------------------------
Running: param
+ RC3_TRIM: unchanged
----------------------------------
Running: param
+ RC4_MAX: unchanged
----------------------------------
Running: param
+ RC4_MIN: unchanged
----------------------------------
Running: param
+ RC4_TRIM: unchanged
----------------------------------
Running: param
+ RC4_REV: unchanged
----------------------------------
Running: param
+ RC6_MAX: unchanged
----------------------------------
Running: param
+ RC6_MIN: unchanged
----------------------------------
Running: param
+ RC6_TRIM: unchanged
----------------------------------
Running: param
+ RC_CHAN_CNT: unchanged
----------------------------------
Running: param
+ RC_MAP_MODE_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_POSCTL_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_RETURN_SW: unchanged
----------------------------------
Running: param
+ MC_PITCHRATE_P: unchanged
----------------------------------
Running: param
+ MC_ROLLRATE_P: unchanged
----------------------------------
Running: mixer
WARN  offboard mission update failed (file /home/roman/Firmware/src/modules/navigator/mission.cpp line 238)
WARN  ERROR: invalid mission state (file /home/roman/Firmware/src/modules/navigator/mission.cpp line 771)
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
INFO  <hilservo> 
INFO  ioctl 0x248f99d0 0x00000000
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
INFO  <hilservo> 
INFO  ioctl 0x248f99d0 0x00000000
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
Enter a command and its args:
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 10134

'''

tumbili, 2015-06-15:
oops

LorenzMeier, 2015-06-15:
I get now the node_open error:
```
WARN  node_open as advertiser failed. (file /home/drone/src/Firmware/src/modules/uORB/uORBManager_posix.cpp line 98)
```

@mcharleb I followed the setup guide, but I think there is still an issue with the orb and with the default network ports, or a major usage failure on my end.

LorenzMeier, 2015-06-18:
@tumbili Now I get a mavlink mission init error, and consequently QGC seems to ignore parts of the system (I don't get to the param editor):

```
Tools/sitl_run.sh
INFO  Shell id is 47412612771520
App name: mainapp
----------------------------------
Running: uorb
----------------------------------
Running: param
----------------------------------
Running: mavlink
arg 0 mavlink
arg 1 start
arg 2 -u
arg 3 14556
WARN  mode: 0, data rate: 2880 B/s on udp port 14556 (file /home/drone/src/Firmware/src/modules/mavlink/mavlink_main.cpp line 1480)
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED
telem> PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
telem> PREFLIGHT FAIL: ACCEL #1 UNCALIBRATED
telem> PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
INFO  Setting up UDP w/port 14556

WARN  offboard mission init: ERROR (file /home/drone/src/Firmware/src/modules/mavlink/mavlink_mission.cpp line 126)
----------------------------------
Running: simulator
arg 0 simulator
arg 1 start
arg 2 -s
----------------------------------
Running: param
+ CAL_GYRO0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC0_ID: unchanged
----------------------------------
Running: param
+ CAL_ACC1_ID: unchanged
----------------------------------
Running: param
+ CAL_MAG0_ID: unchanged
----------------------------------
Running: rgbled
----------------------------------
Running: tone_alarm
INFO  TONE_SET_ALARM 0
----------------------------------
Running: gyrosim
----------------------------------
Running: accelsim
INFO  Simulator started
----------------------------------
Running: barosim
WARN  failed to open /dev/ttyACM0 (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 348)
WARN  failed to send streaming command to /dev/ttyACM0 (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 356)
----------------------------------
Running: adcsim
----------------------------------
Running: gpssim
INFO  <gps> 
INFO  lock
INFO  <gps> 
INFO  unlock
----------------------------------
Running: commander
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 2293760
----------------------------------
Running: sensors
INFO  TONE_SET_ALARM 9
INFO  ToneAlarm::start_tune
WARN  INVAL: ARMING_STATE_STANDBY_ERROR - ARMING_STATE_STANDBY (file /home/drone/src/Firmware/src/modules/commander/state_machine_helper.cpp line 274)
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 2293760
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1376256
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 1310720
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
INFO  IOCTL DEVIOCGDEVICEID 196608
----------------------------------
Running: ekf_att_pos_estimator
.
----------------------------------
Running: mc_pos_control
----------------------------------
Running: mc_att_control
----------------------------------
Running: hil
INFO  <hilservo> 
INFO  MODE_4PWM
INFO  [hilservo] 
INFO  default PWM output device
----------------------------------
Running: dataman
dataman: Unknown restart, data manager file 'rootfs/fs/microsd/dataman' size is 101504 bytes
----------------------------------
Running: param
+ MAV_TYPE: unchanged
----------------------------------
Running: param
+ RC1_MAX: unchanged
----------------------------------
Running: param
+ RC1_MIN: unchanged
----------------------------------
Running: param
+ RC1_TRIM: unchanged
----------------------------------
Running: param
+ RC1_REV: unchanged
----------------------------------
Running: param
+ RC2_MAX: unchanged
----------------------------------
Running: param
+ RC2_MIN: unchanged
----------------------------------
Running: param
  RC2_TRIM: unchanged
----------------------------------
Running: param
+ RC3_MAX: unchanged
----------------------------------
Running: param
+ RC3_MIN: unchanged
----------------------------------
Running: param
+ RC3_TRIM: unchanged
----------------------------------
Running: param
+ RC4_MAX: unchanged
----------------------------------
Running: param
+ RC4_MIN: unchanged
----------------------------------
Running: param
+ RC4_TRIM: unchanged
----------------------------------
Running: param
+ RC4_REV: unchanged
----------------------------------
Running: param
+ RC6_MAX: unchanged
----------------------------------
Running: param
+ RC6_MIN: unchanged
----------------------------------
Running: param
+ RC6_TRIM: unchanged
----------------------------------
Running: param
+ RC_CHAN_CNT: unchanged
----------------------------------
Running: param
+ RC_MAP_MODE_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_POSCTL_SW: unchanged
----------------------------------
Running: param
+ RC_MAP_RETURN_SW: unchanged
----------------------------------
Running: param
+ MC_PITCHRATE_P: unchanged
----------------------------------
Running: param
+ MC_ROLLRATE_P: unchanged
----------------------------------
Running: mixer
INFO  [hilservo] 
INFO  starting
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
INFO  <hilservo> 
INFO  ioctl 0x62ab23a0 0x00000000
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
INFO  <hilservo> 
INFO  ioctl 0x62ab23a0 0x00000000
INFO  <hilservo> 
INFO  lock
INFO  <hilservo> 
INFO  unlock
Enter a command and its args:
INFO  ToneAlarm::start_note 8522
WARN  filter ref off: baro_alt:   0.0000 (file /home/drone/src/Firmware/src/modules/ekf_att_pos_estimator/ekf_att_pos_estimator_main.cpp line 673)
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 8522
INFO  ToneAlarm::start_note 10134
make: *** [sitlrun] Interrupt
```

tumbili, 2015-06-18:
@LorenzMeier I don't see the mavlink error.

closed 2015-06-26 by LorenzMeier