PX4/PX4-Autopilot #2392: "Not yet ready for mission, no position lock" does not emit buzzer warning

tumbili, 2015-06-16:


LorenzMeier, 2015-06-16:
That's not a bug. The buzzer warning will only sound once home is set, because else you will get a warning on every boot, or when you just want to fly manually. It would be a bug if it does not sound after home is set.

tumbili, 2015-06-16:
It depends. This happens after the user has uploaded a mission. So either you want to indicate to him that he cannot fly the mission yet because he doesn't have position lock or you just shut up and assume he reads the info in QGC.

LorenzMeier, 2015-06-16:
I think the "prevent flight when no GPS lock present" is an UI thing in the GCS. We do have a preflight check item which is currently disabled by a circuit breaker. If we flick that to on then things will sequence properly, but users might complain. So it feels more like the mission planning component should ensure it only allows the user to plan once things are initialised correctly.

tumbili, 2015-06-16:
Well, completely not to allow him planning also doesn't make sense because he might want to plane the mission at home.

LorenzMeier, 2015-06-16:
Logged here:
https://github.com/mavlink/qgroundcontrol/issues/1667

closed 2015-06-16 by LorenzMeier