PX4/PX4-Autopilot #2393: Multicopter position control: Attitude control oscillation

LorenzMeier, 2015-06-16:
Summarising some comments from @SimonWilks here:

Logs:
http://dash.oznet.ch/view/SHgcgQZ7wcLX9DQV6B7pMU
http://dash.oznet.ch/view/n6psxmgeH3EXg4JntrPH6M

The trigger is the combined low-then-high throttle event, which leaves the system for some time without effective attitude control:
http://dash.oznet.ch/view/SHgcgQZ7wcLX9DQV6B7pMU#Thrust_PLOT

I suspect this code path could play a role in this, as its defeating the new safety built into the mixer. We probably should completely remote it, as it can lead to a loss of attitude control.
https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1204-L1212

DrTon, 2015-06-16:
>https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1204-L1212

This code only reduces roll/pitch of the copter in POSCTL in case if at full thrust copter can't hold altitude. Not related to this issue.

I've looked this log and see the following effects:
- at 57.72 pitch setpoint increased to start breaking
- at 57.85 pitchrate was already big enough and attitude controller reversed pitch control to stop pitch motion
- but at 57.9 copter was already climbing up with significant velocity (because props are working also like wings, and give vertical thrust even without thrust), so altitude controller reduced throttle (note, that it was low-end saturation, not high-end that noted by Lorenz)
- even with mixer hacks pitch control was not strong enough at this thrust to stop quickly.
- copter has significant asymmetry, at hovering thrust for motors 1,2 is more than 3,4, because of this copter got also some yaw movement, and after some rotation around body Z axis got significant roll. Mixer hacks helped to leave yaw without control.

My opinion what is wrong here:
- roll/pitch/yaw gains are maybe a bit too high
- the worst thing is MPC_Z_VEL_P=0.3, it's 3 times higher than default value, as result altitude controller goes into saturation, recent mixer hacks help to mask this but not always. As I understand you want to get super precise altitude control, but even overpowered copter can descend only with 1g acceleration ;) so you will get only mixer saturation and and all side effects.
note that if mixer goes into saturation on some aggressive movement you will always have bad performance, masking it with mixer hacks is not good idea IMO
- recent mixer changes only mask one problem and creating another, and make behavior more unpredictable, bad attitude control problems should be solved another way.

tumbili, 2015-06-16:
One of the reason why the thrust setpoint went down that far was because the attitude setpoint more or less jumped to a new setpoint. So now the desired and the actual thrust vector were subject to a large angle offset. I also had this problem a lot when writing the vtol controller for the tailsitter. In that case the effect was even worse because a low throttle setting lead to a complete loss of pitch control.
I solved the problem by limiting the attitude error.

tumbili, 2015-06-16:
[image]

tumbili, 2015-06-16:
[image]

tumbili, 2015-06-16:
Green is SP glube is actual. The time difference between those two snapshots is just 0.03 s

SimonWilks, 2015-06-16:
@DrTon yes, yaw still needs tuning. I'll take a look at that. I'd tried with lower pitch/roll gains already and pushed them up as hard as I could hoping it would help recover faster. In manual it flies very well so it's certainly capable. MPC_Z_VEL_P we increased to improve it's altitude recovery which improved significantly then, but I can reduce it again to see if it plays a role here.

@tumbili that seems to reflect what I saw (obviously:) and that the attempts to correct are so sudden that it is floundering in the air...could there be a kind of prop stall in this situation? That's how it looks.

tumbili, 2015-06-16:
@SimonWilks I think we all agree that with this low thrust setpoint the copter would have never been able to recover. I'll look into this and see what we can do.

LorenzMeier, 2015-06-16:
This boils down to the correct vs. robust argument. We want to be correct, but we also need to be robust. Tuning the gains to the right values will make us correct, but not robust to either disturbances (we have seen enough corner cases in logs) or suboptimal tuning.

closed 2016-01-06 by LorenzMeier