PX4/PX4-Autopilot #2399: Well defined sys logging

davids5, 2015-06-16:
We need a way to define system logging like Log[j|net]  that is platform independent and does not have idefs in every file that uses it.

This will also centralize the changes for the version of nuttx on master vr the latest upstream used in nuttx next.

Proposed - with todos that need resolution
```
/****************************************************************************
 *
 *   Copyright (C) 2015 PX4 Development Team. All rights reserved.
 *	 Author: David Sidrane <david_s5@nscdg.com>
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions
 * are met:
 *
 * 1. Redistributions of source code must retain the above copyright
 *    notice, this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright
 *    notice, this list of conditions and the following disclaimer in
 *    the documentation and/or other materials provided with the
 *    distribution.
 * 3. Neither the name PX4 nor the names of its contributors may be
 *    used to endorse or promote products derived from this software
 *    without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS
 * "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS
 * FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE
 * COPYRIGHT OWNER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT,
 * INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING,
 * BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS
 * OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED
 * AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT
 * LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN
 * ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE
 * POSSIBILITY OF SUCH DAMAGE.
 *
 ****************************************************************************/
#ifndef PX4LOG_H_
#define PX4LOG_H_

#include "px4_config.h"

#include <stdint.h>
#include <stddef.h>

//todo:RENAME NO_BLOAT to PX4_CONFIG_NO_LOG_FILE_NAMES
//todo:define px4_panic
#include "px4_panic.h"
#include "px4_macros.h"
//todo:Position px4_macros.h in tree
//todo:Define Notion of DEBUG_BUILD and RELEASE_BUILD

/*
 * This module supports runtime and compile time message filtering.
 *
 * For a message be compiled in the message type has to be >= to 
 * then the PX4_LOG_LEVEL_AT_COMPILE_TIME
 * For a message to be output the message type has to be >= 
 * then the PX4_LOG_LEVEL_AT_RUN_TIME
 *
 * This module assumes one of two 
 * #defines RELEASE_BUILD or DEBUG_BUILD exist
 *
 * Then the PX4_LOG_LEVEL_AT_COMPILE_TIME and PX4_LOG_LEVEL_AT_RUN_TIME are 
 * set based on RELEASE_BUILD or DEBUG_BUILD
 * For a release build
 *     PX4_LOG_LEVEL_AT_COMPILE_TIME = PX4_WARN_LEVEL
 *     PX4_LOG_LEVEL_AT_RUN_TIME 		=  PX4_WARN_LEVEL
 * For a debug build:
 *     PX4_LOG_LEVEL_AT_COMPILE_TIME = PX4_LOG_LEVEL
 *     PX4_LOG_LEVEL_AT_RUN_TIME 		 = PX4_LOG_LEVEL
 *
 * The API
 * PX4_LOG(fmt,...)
 * PX4_DEBUG(fmt,...)
 * PX4_WARN(fmt,...)
 * PX4_ERROR(fmt,...)
 * PX4_PANIC(code,fmt,...)
 */

// Debug Levels
#define PX4_LOG_LEVEL       1
#define PX4_DEBUG_LEVEL     2
#define PX4_WARN_LEVEL      3
#define PX4_ERROR_LEVEL     4
#define PX4_PANIC_LEVEL     5

#if !defined(INCLUDE_FILE_INFO_IN_DEBUG) // See px4_macros.h for NO_BLOAT
#define _PX4__FILE_PATH          _FILE_NAME_ 
#else
#define _PX4_FILE_PATH          NULL
#endif  //

#if defined(DEBUG_BUILD)
#define PX4_LOG_LEVEL_AT_COMPILE_TIME PX4_LOG_LEVEL
#define PX4_LOG_LEVEL_AT_RUN_TIME 		PX4_LOG_LEVEL         
#endif

#if defined(RELEASE_BUILD)
#define PX4_LOG_LEVEL_AT_COMPILE_TIME PX4_ERROR_LEVEL
#define PX4_LOG_LEVEL_AT_RUN_TIME 		PX4_ERROR_LEVEL
#endif

#ifdef __cplusplus
extern "C" {
#endif
// Must be provided by platform
extern uint32_t px4_log_level_at_run_time __attribute__ ((weak));
void px4_log_print_(int level, int line, const char *func, const char *file, const char *msg, ...);
void px4_panic_(int code);
#ifdef __cplusplus
}
#endif

// Short Cuts
#define __PX4_LOG_LEVEL_TEST(level) (level >= PX4_LOG_LEVEL_AT_COMPILE_TIME && level >= PX4_LOG_LEVEL_AT_RUN_TIME)

#if defined(PX4_CONFIG_USE_ONLY_PANIC)
#define PX4_LOG(fmt, ...)
#define PX4_DEBUG(fmt, ...)
#define PX4_WARN(fmt, ...)
#define PX4_ERROR(fmt, ...)
#define PX4_PANIC(code,fmt, ...) do {px4_panic_(code);}while(0)
#else
// Macros to use
#define PX4_LOG(fmt, ...)    do { if ( __PX4_LOG_LEVEL_TEST(PX4_LOG_LEVEL)  )  {px4_log_print_(PX4_LOG_LEVEL,__LINE__,__PRETTY_FUNCTION__,_FILE_PATH,fmt, ##__VA_ARGS__);}}while(0)
#define PX4_DEBUG(fmt, ...)  do { if ( __PX4_LOG_LEVEL_TEST(PX4_DEBUG_LEVEL))  {px4_log_print_(PX4_DEBUG_LEVEL,__LINE__,__PRETTY_FUNCTION__,_FILE_PATH,fmt,##__VA_ARGS__);}}while(0)
#define PX4_WARN(fmt, ...)   do { if ( __PX4_LOG_LEVEL_TEST(PX4_WARN_LEVEL) )  {px4_log_print_(PX4_WARN_LEVEL,__LINE__,__PRETTY_FUNCTION__,_FILE_PATH,fmt,##__VA_ARGS__);}}while(0)
#define PX4_ERROR(fmt, ...)  do { if ( __PX4_LOG_LEVEL_TEST(PX4_ERROR_LEVEL) ) {px4_log_print_(PX4_ERROR_LEVEL,__LINE__,__PRETTY_FUNCTION__,_FILE_PATH,fmt,##__VA_ARGS__);}}while(0)
#define PX4_PANIC(code,fmt, ...)  do { if ( __PX4_LOG_LEVEL_TEST(PX4_PANIC_LEVEL) ) {px4_log_print_(PX4_PANIC_LEVEL,__LINE__,__PRETTY_FUNCTION__,_FILE_PATH,fmt,##__VA_ARGS__);} px4_panic_(code);}while(0)
#endif

#endif /* PX4LOG_H_ */

```

davids5, 2015-06-16:
@mcharleb 

This is still 1/2 baked, but I want to run it by you to get your requirements and suggestions. We can have a skype call if need be.

David

mcharleb, 2015-06-16:
We also need a PX4_INFO level that is not compile time filtered for output messages such as: list_devices(), dump_counters(), etc whose whole purpose is to provide output.

LorenzMeier, 2015-07-03:
Closing this as the one from @mcharleb is merged.

closed 2015-07-03 by LorenzMeier