PX4/PX4-Autopilot #2407: FMUv  does not show voltage (stable)

bansiesta, 2015-06-17:
The battery voltage does not work anymore on FMUv1 (stable branch).

The ADC channel is read in https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L1709 but the voltage read is 0.01v to 0.02v and garbage. It's not a param scaling issue.

The FMU is used on an IO and the battery port is connected.

LorenzMeier, 2015-06-18:
The battery voltage is read by IO for FMUv1. Are you sure you sure you're looking in the right place? Please run ```px4io status```. It never was in the FMU driver on v1. Closing this issue.

bansiesta, 2015-06-18:
```
vbatt mV 292 ibatt mV 0 vbatt scale 10000
```
So you're saying this was never working?

bansiesta, 2015-06-18:
@LorenzMeier Has this worked on FMUv1 before or not?

LorenzMeier, 2015-06-18:
It should work in the default config. Is this a regression or a hardware issue? @tumbili Does your FMUv1 setup work with battery voltage?

tumbili, 2015-06-18:
Works for me, see e.g. this log  http://dash.oznet.ch/view/Eo3MzgwfnGZCiK9QJB9Q44

tumbili, 2015-06-18:
This is release though, don't know about master.

bansiesta, 2015-06-19:
I think what happens is that the scaling param is not applied on the IO side. I had to manually change the scaling on the IO side to make it work. I did try thr BAT_V_SCALE param but it didn't have any effect.

LorenzMeier, 2015-06-19:
The scaling on IO is a fixed number, it never ever gets modified. This suggests you have a hardware issue.

closed 2015-06-19 by LorenzMeier

bansiesta, 2015-06-28:
Swapped the IO board, yes it was a hardware issue! Sorry and thanks!