PX4/PX4-Autopilot #2411: Re-check mag calibration for external magnetometers

LorenzMeier, 2015-06-18:
We have two reports where the external magnetometer did not calibrate correctly.

LorenzMeier, 2015-06-19:
@mhkabir I could not reproduce your issue. Could you describe the issue you were having?

mhkabir, 2015-06-19:
I have my external UAVCAN mag as MAG0 and the internal one as MAG1 (using the startup script timing hack) 
On attempting calibration, the first calibration goes OK ( QGC displays graphical steps, etc.) next, it asks for external mag orientation. On choosing the rotation and clicking "OK", the right panel slides out and there is no more further information shown in QGC other than "calibration successful" ( in reality, the flight controller has gone into calibration of the second mag, and shows messages in the mavlink console)

LorenzMeier, 2015-06-19:
Are you sure you are using a recent Firmware version? All mags are calibrated at once, unless you use something which is many weeks old.

mhkabir, 2015-06-19:
I'm using this : https://github.com/mhkabir/Firmware/tree/artemis
Which is pretty new.

LorenzMeier, 2015-06-19:
Ok, could you share the console output? I have looked into this but I don't have a starting point to understand what / how it fails.

LorenzMeier, 2015-06-22:
I'm closing this now since we have no reproductions of the initial report.

closed 2015-06-22 by LorenzMeier