PX4/PX4-Autopilot #2413: mad yaw rotation

taogashi, 2015-06-18:
Hi all,
     My SingleCopter experienced a mad rotation around z axis, I've checked the log and found something interesting, but still don't know the cause of this behavior. 
[image]
As the legend shows, yaw_rate_setpoint, attitude_controller_output_yaw, yaw_rate and rc_ch3 are plotted. There is an apparent rising of yaw_rate, which is not intended.

a normal log looks like:
[image]

I think there is a considerable latency between yaw_rate and attitude_controller_output_yaw. But I don't think it's the cause of the mad rotation.

DrTon, 2015-06-18:
Maybe wrong props rotation direction? Post raw log please.

taogashi, 2015-06-18:
Hi @DrTon raw log is [here](https://github.com/taogashi/dragon)

taogashi, 2015-06-18:
log001.px4log is the mag one, log002.px4log the normal, :)

LorenzMeier, 2015-07-03:
@taogashi When you refer to "SingleCopter" what do you mean by that? Does it have single rotor?

taogashi, 2015-07-13:
@LorenzMeier ducted fan VTOL actually. The bug can't be reproduced... we are taking more analyses to find the answer. Thx

LorenzMeier, 2016-12-26:
Stale, closing.

closed 2016-12-26 by LorenzMeier