PX4/PX4-Autopilot #2419: Failing compilation on eigen test

TSC21, 2015-06-18:
I updated eigen lib to latest version: `Subproject commit 9aed5e67812dd449ddea40749e65331813360fde`. Now I'm getting the following when compiling:
```bash
In file included from /home/nuno/dev_px4/landing_target_support/Firmware/src/lib/eigen/Eigen/Core:296:0,
                 from /home/nuno/dev_px4/landing_target_support/Firmware/src/lib/px4_eigen.h:49,
                 from /home/nuno/dev_px4/landing_target_support/Firmware/src/systemcmds/tests/test_eigen.cpp:41:
/home/nuno/dev_px4/landing_target_support/Firmware/src/lib/eigen/Eigen/src/Core/Matrix.h:11:0: error: unterminated #ifndef
 #ifndef EIGEN_MATRIX_H
 ^
compilation terminated due to -Wfatal-errors.
```

LorenzMeier, 2015-06-19:
An upgrade and fix would be appreciated, I just want to set expectations right for the time budget we have available for this currently.

If you update Eigen its within your responsibility to fix any update breakage - is there a specific reason to upgrade the version? We can only allocate bandwidth for bugfixes on Eigen right now, and we will do an Eigen upgrade when needed for the general development - which, unless you have new information, is not now.

TSC21, 2015-06-20:
I will take a look on this. It just updated cause I issued that all submodules got updated to latest commit, when I was just looking for updating only mavlink. But then we don't loose anything updating the modules, so I'll see what's the problem.

TSC21, 2015-06-22:
It was a merging problem. Closing now

closed 2015-06-22 by TSC21