PX4/PX4-Autopilot #2420: Testing the Q attitude estimator in coordinated turns

kd0aij, 2015-06-18:
I have observed on both a 250 and 450 quad that flying in a circle with bank angle and yaw rate to approximate a fixed-wing "coordinated turn" results in error in the attitude estimate. When roll and pitch are returned to neutral, the copter is noticeably tilted. The tilt is noticeable after 3 or 4 circuits.
After circling, I hold the copter level (very little wind during the test), so roll and pitch should be near zero if the attitude estimate is correct.
Q estimator log: https://copy.com/3RS92YD4mlih7EyF

I was unable to demonstrate a similar attitude error using the EKF attitude estimator.
EKF estimator log: https://copy.com/AhVSyBftYqx6iX9e

tumbili, 2015-06-19:
@kd0aij Very interesting! Could you use the level horizon option in QGC on a level surface, fly the manoeuvre again and after that put it on the level surface one more and observe the attitude.

kd0aij, 2015-06-19:
@tumbili This is a well-understood dynamic error in the complementary filter, caused by the centripetal acceleration in a coordinated turn. You can see that the attitude error decays to zero over about 20 seconds in this snapshot, but looking at the magnitude of the pitch correction I'm using to maintain a level attitude:
[image]

Bill Premerlani implemented compensation for this error in the MatrixPilot IMU.
Bill's algorithm originally relied on 3D velocity from GPS, but after seeing what Horizon Hobby was doing with their non-GPS "SAFE" system he looked at the problem again: http://diydrones.com/forum/topics/centrifugal-compensation-with-only-gyros-and-accelerometers?xg_source=activity

LorenzMeier, 2015-06-19:
The filter is so short it might make sense to sift through it. It does compensate using GPS, there is probably something not quite right in how GPS is read out.

kd0aij, 2015-06-19:
Verified that ATT_ACC_COMP parameter is > 0 in the log, but it looks like gpos.eph was too high, so accel comp was not applied.

kd0aij, 2015-06-19:
I'll have to repeat this test with GPS. I thought I waited for a green LED before arming each time, but the Q test was the first session, so maybe I forgot the first time.

kd0aij, 2015-06-19:
Repeat of Q test with GPS fix eliminated the attitude error accumulation during coordinated turns.

closed 2015-06-19 by kd0aij