PX4/PX4-Autopilot #2430: POSIX: Sensor sim produces imperfect data

LorenzMeier, 2015-06-20:
@mcharleb The data we get from the sim must be distorted in time, scale or axis assignment. The performance is significantly worse compared to HIL.

tumbili, 2015-06-22:
@LorenzMeier Will look into this.

LorenzMeier, 2015-06-22:
A test shows that this only affects the dynamic attitude estimation: The system interprets ~1.5 radians as 0.6 radians, so it systematically underestimates the real rotation. This is either a time delta issue or a scaling issue of the converted values.

tumbili, 2015-06-23:
@LorenzMeier The simulated drivers are using a timestamp which they generate themselves. I will replace that one with the timestamp of the HIL sensor message.

tumbili, 2015-06-23:
Also I checked the interval in which data is coming from the simulator and got on average 3ms.

LorenzMeier, 2015-06-24:
The stopgap fix was this one:
https://github.com/PX4/Firmware/pull/2455

But the real fix is to block the command handler until the thread it launched returned.

tumbili, 2015-06-25:
Related to #2465

LorenzMeier, 2015-06-26:
@mcharleb I've traced this now to the HRT implementation and/or the gyro driver not delaying the sensor reads as much as it should. the inter-loop times are at 1 KHz, but should be at 250 Hz:
```
WARN  last run: 813 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 883 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 980 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 956 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 133 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 511 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 784 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
WARN  last run: 685 (file /home/drone/src/Firmware/src/modules/sensors/sensors.cpp line 2201)
```

mcharleb, 2015-07-02:
This should have been fixed with the latest batch of changes. The code that did the limiting to 250Hz was ifdefed out for the posix build. I tested it and it seems to work for me.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier