PX4/PX4-Autopilot #2444: MS5611 on I2C PXFMUv1

Janesak1977, 2015-06-22:
In file ms5611_i2c.cpp, line 165 and line 75, function name dev_ioctl in declaration and implementation of IOCTL function is not good. Name must be ioct.
MS5611 driver is not working on PX4FMUv1 (MS5611 is on I2C), because of that declaration.

LorenzMeier, 2015-06-22:
Could you send the patch for the fix? I don't understand what you exactly mean without a reference to the code changes.

Janesak1977, 2015-06-22:
Here is patched ms5611_i2c.cpp.file.
See declaration and implementation of function ioctl.

LorenzMeier, 2015-06-22:
Could you please attach the file in the Guthub web form or copy its content there? You can't attach files via email to github discussions.

kd0aij, 2015-06-22:
While the baro seems to be working on my FMUv1 running the release_v1.0.0 branch, ALTCTL does not. As soon as I switch to ALTCTL, my QAV250 climbs aggressively. Log: https://copy.com/myYzZB2oYGNPYqtt
Snapshot: 
[image]

Janesak1977, 2015-06-22:
Yes, in release_v1.0.0 branch it is OK. 
Master branch declaration:
	virtual int	dev_ioctl(unsigned operation, unsigned &arg);
release_v1.0.0 branch declaration
	virtual int ioctl(unsigned operation, unsigned &arg);

kd0aij, 2015-06-22:
This (regression?) is from commit cd30b4d5caf717d77b5f5d30ba696408a6062387 by Mark Charlebois. Something to do with posix virtual devices?

Janesak1977, 2015-06-22:
Initially it is from commit https://github.com/PX4/Firmware/commit/d013ac092769f7975e540a24c5756002251248a4 by @mcharleb

LorenzMeier, 2015-06-27:
Fixed via 5e27b7b, thanks @tridge.

closed 2015-06-27 by LorenzMeier