PX4/PX4-Autopilot #2445: bad IMU sensors in FMUv1 after crash?

kd0aij, 2015-06-22:
My QAV250 isn't flying well since a crash into a tree. Since the FMU/IO stack is hard mounted to the frame, I suspect that the accelerometers and/or the gyros have been damaged. While calibration completes without errors and the accel/gyro offsets look reasonable, the copter doesn't feel as controllable as before, and develops fairly large errors in roll and pitch when maneuvering. Yaw is also tending to drift now, it does not hold heading very well.
Here's a log showing both manual mode flight and ALTCTL. ALTCTL is unusable now, on first entering that mode, the copter climbs aggressively about 5-10m, and response to the throttle stick is so unpredictable that is not possible to set a particular altitude:
https://copy.com/9zPYGAI8GmERZgMu

kd0aij, 2015-06-22:
Wait Wait: that flight was with internal compass and ekf_att_pos_estimator. INAV_ENABLED was inadvertently set to zero. Need to fly again with the Q estimator.

LorenzMeier, 2015-06-22:
You will want this branch if you want to give the ekf_att_pos_estimator a ride:
https://github.com/PX4/Firmware/pull/2438

kd0aij, 2015-06-22:
I think my Z accelerometer my be broken:
[image]

kd0aij, 2015-06-22:
I don't suppose a bent motor shaft or out-of-balance propellers would show up only in Z?

kd0aij, 2015-06-23:
@AndreasAntener I think a collision with a tree trunk broke my Z accelerometer. Link to log and a plot are in the comments above; do you think I've come to the right conclusion?

LorenzMeier, 2015-06-24:
It looks broken - closing this.

closed 2015-06-24 by LorenzMeier