PX4/PX4-Autopilot #2457: -Wno-unused-variable: INAV estimator

yyassa, 2015-06-24:
In a recent commit (736125441ecb4) in master, the following flag was added to allow unused variables in INAV estimator temporarily

In src/modules/position_estimator_inav/module.mk line45:
```
EXTRACFLAGS = -Wframe-larger-than=3500 -Wno-unused
```

However i still could not compile master. This is solved if you replace it by:
```
EXTRACFLAGS = -Wframe-larger-than=3500 -Wno-unused-variable
```

(also checked the gcc documentation and only found '-Wno-unused-variable' and no '-Wno-unused')

Could this have been an unseen bug or does it depend on the gcc version or similar?

LorenzMeier, 2015-06-24:
Its not a bug but depends on the GCC version. The right fix is to remove the flag and comment out the offending print / log / debug variable.

yyassa, 2015-06-24:
Alright, thanks for the info! 
Should I fix this? or do you mean to leave the debugging commented temporarily for a reason?

closed 2016-05-22 by LorenzMeier