PX4/PX4-Autopilot #2463: Add Model Checking Support

jgoppert, 2015-06-25:
Ensuring reliability between commits is crucial and difficult to do with so many developers, so I have been working on adding model checking support. In addition, this will most likely be a requirement to certify many drones for commercial use.  I have a small proof of concept repo here: https://github.com/jgoppert/px4vv 

See the README for an example of a state machine verification that closely follows the PX4 commander code.

I'm looking for developers who are interested in  helping me get a verified version of the PX4 codebase up and running. We can start small and simple for now, for example, only support quadrotors. I also would like to keep the build system in cmake if possible.

LorenzMeier, 2015-06-27:
That sounds great, and is something I have been wanting to do forever! But to prevent this becoming another unmaintained great idea it needs to scale, in particular to the new hardware platforms we're bringing up.

I think its a great initiative like a few you've proposed in the past already, but I want to make sure it doesn't end up as abandoned, unmaintained one-shot experiment, and for that to not happen its crucial this becomes the default. If you just want to propose it as "safe alternative" you would need to put up a full-time position in order to keep it up to date with the main project. Unless that's the plan it is critical to move over our complete system to the model checked version over time.

For that to be possible, you will need to compromise and be pragmatic:

  * You need to make sure that SITL is working, and in CMake its not. So either you convert the complete state on master in a fashion that is supported by the dev community (getting buy-in from the core devs, including Qualcomm), or you should just drop CMake. It is completely unrelated to model checking and linking the two topics is greatly complicating the proposal. Model checking without good SITL makes no sense. Also keep in mind that CMake might not be available on all platforms we care about.

  * You will not be able to maintain this and others will not be able to contribute if its not becoming the default. So it needs to become the default state machine and running the checker needs to be part of the main docs.

jgoppert, 2015-06-27:
**Adoption**
 * I agree that maintaining two versions isn't reasonable and we should try to get it adopted as soon as possible.
 * I think you are correct, that the easiest path for adoption is using the existing build system. For now I might leave my test setup in CMake, but when I pull it into a firmware version I will just compile with the existing build system.

**Plan**
 * I will try to do this module by module, for instance, just replacing the commander by a verified version.
 * CBMC: I'm carefully reading the CBMC manual to understand how to embed assumptions etc. in the  code that make verifying more complex code possible. Also, they recommend using SatAbs in combination with CBMC so I'm reading up on that as well.
 * NuttX: I'm just going to abstract all OS calls and ignore NuttX for now.
 * commander: This is the most obvious place that model checking can help and my first priority.
 * uORB: I would like to verify the uORB interaction, this will require rewriting uORB to be in a verifiable form, simple C code with OS specifics abstracted.
 * mavlink: It would be nice to verify the mavlink communcation with the ground station.

**Plan**
**Help**
 * If anyone else is interested, let me know.

LorenzMeier, 2015-06-27:
When you change the commander, please feel free to C++ify it as wanted / needed and/or do a cleanup. Its unreadable right now and the rewrite has been on the list for ages. I'll try to find more people to help with this, as I think its really a great initiative!

jgoppert, 2015-07-01:
Here is the start of the code:
https://github.com/jgoppert/Firmware/tree/commander2/src/modules/commander2

Let me know what you think before I get too far into it.
I just embedded a cmake file to do the testing.

jgoppert, 2015-07-01:
I've decided to go with this: https://en.wikipedia.org/wiki/State_pattern for the finite state machine. It is very organized and extensible while the table felt very cramped and brittle.

jgoppert, 2015-07-01:
Also, I haven't gotten very far yet, here is  a more complete example of the state pattern style fsm https://github.com/jgoppert/vvpilot/tree/master/src/vvp/modules/commander

LorenzMeier, 2015-07-01:
The pattern itself sounds very useful. We did that in the navigator app, and the intent was to rewrite the commander in a similar fashion:
https://github.com/PX4/Firmware/tree/master/src/modules/navigator

closed 2018-02-22 by PX4BuildBot