PX4/PX4-Autopilot #2474: MC: Increased thrust problem

tumbili, 2015-06-26:
@sjwilks @AndreasAntener I opened this issue so that we don't forget. Could you provide some explanation (and logs) what the problem is?

SimonWilks, 2015-06-26:
There are two things we noticed. 

Firstly, it is possibly simply an issue of getting the right default value for MPC_THR_MIN. A default of 0.1 might work for standard multirotors but the problem is that for some it will result in lift off on arming or tipping over when the props don't spin up at the same time. On the TBS Long Range even 0.05 was still too high and we went for 0.01 but didn't test intermediate values. On a home-made VTOL quad it took a while to get it down to the ground and commanding yaw seemed to have little effect. So the question becomes: what is less safe, a multirotor that launches on arming (and will be hard to bring back down) or loss of stability on descent. I would vote for a low default and document how to improve performance in a tuning guide.

Secondly, there was a strange effect on the FireFly when switching back from FW to MC where the min. throttle seemed to not be applied and some of the motors were not spinning for a second or two, however, it's likely this might be more to do with VTOL transition logic. The problem went away when setting MC_THR_MIN to 0.

LorenzMeier, 2015-06-27:
What you're describing sounds rather like an ESC calibration problem. With a lower value of MPC_THR_MIN the "spread" between motors is lower, resulting in higher PWM values for the lower spinning motors. Could it be that you need to run an ESC calibration on both platforms?

SimonWilks, 2015-06-27:
Actually that could be an issue. We did re-run the ESC calibration and noticed that throttle was even higher than before. I recall a calibration issue being discussed at some point recently, could it be related?

LorenzMeier, 2015-07-03:
Fixed on all relevant branches.

closed 2015-07-03 by LorenzMeier