PX4/PX4-Autopilot #2476: Tuning Gains EKF

boosfelm, 2015-06-26:
As discussed posting our tuning values for Sensor and Bias noise, which led to much better estimation results. Thanks @surberj for testing this. It has to be noted that this can be really aircraft specific so please be catious when testing. In our case the Pixhawk is embedded in a foam structure of a VTOL tailsitter design.

PE_ABIAS_PNOISE = 1.0e-5
PE_ACC_PNOISE = 0.05
PE_POSD_NOISE = 1.25 (estimator diverged at PE_POSD_NOISE = 3.0)
PE_VELD_NOISE = 0.3 (estimator diverged at PE_VELD_NOISE = 0.05)

According to the data PE_POSD_NOISE = 2.0 and PE_VELD_NOISE = 0.15 look best, but we have not enough evidence. I would rather choose a trade-off between enhancing and staying near the working solution (which surely does not diverge), i.e,
PE_POSD_NOISE = 1.25
PE_VELD_NOISE = 0.3

This might also help on #1902.

LorenzMeier, 2015-06-26:
Thanks! I took your suggestion into this PR:
https://github.com/PX4/Firmware/pull/2478

LorenzMeier, 2015-07-03:
Since this is merged, closing, thanks!

closed 2015-07-03 by LorenzMeier

boosfelm, 2015-07-04:
great, happy to contribute. Yesterday we did 4 flights about 10-15 min each with master_release (+ our own attitude control module) in different modes, namely fw att control, fw altitude control, fw position control, loiter, mission and LPOS_Z was off by 0.2 to 1.9 m.