PX4/PX4-Autopilot #2479: TeraRanger One slow boot

lfrodrigues, 2015-06-26:
The teraranger one driver in PX4 sometimes does not work correctly:
When the sensor is turned on at the same time as pixhawk sometimes driver will fail to probe it (it takes a bit for the sensor to be ready).
If I connect pixhawk after I connect the sensor to power all works fine. 

I think the simplest solution is to add a 1 or 2 seconds delay when the driver is loaded. Is there a preferred way to make a delay in the driver's probe function?

LorenzMeier, 2015-06-27:
As long as the delay does not introduce a busy loop of some sort its welcome.

lfrodrigues, 2015-06-27:
Ok I see. Is there some example code in other part of the firmware do know which API to use?

LorenzMeier, 2015-06-27:
Not yet. The right pattern would be to start the driver normally (register the work queue entry) and then do the probe in there until it succeeds or times out. So like reading the sensor, but instead you just keep probing it in 500 ms intervals or so.

lfrodrigues, 2015-07-09:
We updated the firmware, no longer have this problem.

Thanks for your comment Lorenz!

closed 2015-07-09 by lfrodrigues

mzahana, 2016-11-27:
So, is teraRanger one compatible with PX4 now?

LorenzMeier, 2016-11-27:
@mzahana If an issue is closed it means its fixed. So yes.

lfrodrigues, 2016-11-27:
@mzahana  It should work out of the box, if not please contact TeraRanger support and we will be happy to help you.

mzahana, 2016-11-28:
@LorenzMeier Allow me to disagree. If an issue is closed, it means it worked with at least the people who were able to solve it :)
Also, there is no clear documentation from PX4 side. Thanks anyway.

@lfrodrigues What kind of wiring does it require to work out-of-the-box? Are there any settings that should be done from Pixhawk side? e.g. in QGC?
Thanks.

lfrodrigues, 2016-11-29:
@mzahana TROne has been working in qgroundcontrol since #5359 was merged. 

To make it work you have to:
1. Download QGroundControl from http://qgroundcontrol.com/downloads/
2. Click parameters
3. search trone
4. put 1 in value and click save
5. reboot vehicle and all should be working 

For physical connection see here: http://ardupilot.org/copter/docs/common-teraranger-one-rangefinder.html

mzahana, 2016-11-30:
Thanks. I will for the I2C adapter and try it out.