PX4/PX4-Autopilot #2480: DISTANCE_SENSOR multi-subscribe still broken

mhkabir, 2015-06-27:
I've been trying to get the PX4Flow sonar + LL40LS working together. 

```ls /dev/``` gives me :
```
 range_finder0
 range_finder1
```
I can see that the PX4flow sonar gets instance ID 0 ( Can see data in QGC )
I assume that the ll40ls is getting uorb instance ID 1, but using orb_subscribe_multi() with the instance set to 1 doesn't give me any data.

Any insights?

LorenzMeier, 2015-06-27:
Could you send a link to the exact line you're using? I'm pretty confident its not broken and if it is I would love to understand how to reproduce it.

Thanks!

mhkabir, 2015-06-28:
Subscribing via the wrapper here : https://github.com/PX4/Firmware/compare/master...mhkabir:artemis#diff-478580be4572c29aaaa7e5861078259eR31
Instance ID is the third argument in the ctor.

Wrapper simply calls orb_subscribe_multi here : https://github.com/PX4/Firmware/compare/master...mhkabir:artemis#diff-c036ce2c8ca0551dfec52f22f8578beaR86

AndreasAntener, 2015-09-02:
I hooked up the Lidar (PWM) and Flow. In my case, only Lidar distance gets streamed via Mavlink. If I disable the Lidar, sonar distance gets streamed. The mavlink messages are using `orb_subscribe_multi` as far as I can see.

AndreasAntener, 2015-10-29:
I believe this is fixed.

closed 2015-10-29 by AndreasAntener