PX4/PX4-Autopilot #2486: Auto-generation of uORB -> log format

LorenzMeier, 2015-06-28:
We use a Python [EmPy](https://pypi.python.org/pypi/EmPy/3.1/) based code generation setup for generating the C-code to do inter-process-communication using uORB.

Links to the implementation:
  * [templates](https://github.com/PX4/Firmware/tree/master/msg/templates/uorb)
  * [messages](https://github.com/PX4/Firmware/tree/master/msg)

Now what we would like to do is to not only generate the uORB headers, but also the code to log these to the microSD. There are a few things that we would like to change in the log format in the process.

  * The current maximum field number is 16. We should remove any limitations there.
  * The current maximum description string length is 64. We should also remove any limitations there.
  * Currently not each message has its own timestamp. We would like to change this so that each message is timestamped individually.

Here are the steps required to build this:
  * Modify the sdlog2 app and convert it to C++.
  * When auto-generating, we would not require separate structs for the log content, so these could all be removed: [log structs](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h#L51-L67)
  * The entries here should be auto-generated from the msg spec: [log format spec](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2_messages.h#L506)
  * The header inclusion should be generated: [headers](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L65)
  * The buffer union (or a void pointer struct of sufficient size) should be auto-generated: [buffer](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1060-L1095)
  * Writing the headers should be auto-generated: [header write](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L767-L786)
  * The copy operations from uORB to log buffer should be generated for all topics: [copy ops](https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L1307-L1317)
  * Each uORB message spec should have an entry (a magic variable named e.g. _logging_rate) defining the maximum logging rate
  * Add a mechanism to add a mapping from a field name to a standard label (e.g. ATT.xrate = dronecode:rates_phi) to allow a standardised plotting tool to rely on standard labels

The final goal is that by simply adding a new uORB topic you automatically have the same variable in the log file.

kd0aij, 2015-06-30:
Since I'd like individually timestamped log entries to do impulse response estimation, I'd like to help out on this, but my Python is a bit rusty, and also not very "Pythonic"...

DrTon, 2015-07-13:
>The copy operations from uORB to log buffer should be generated for all topics: copy ops

No need for this, ideally it would be good to use uORB struct as log buffer. But the problem is aligning, for logging it's better to pack all fields. Need to use packed structs for topics or auto-generate unpacked->packed copying function for each topic.

>Add a mechanism to add a mapping from a field name to a standard label (e.g. ATT.xrate = dronecode:rates_phi) to allow a standardised plotting tool to rely on standard labels

"Standartized" labels are not more than "somebody wrote short ugly field name because of lack of space in APM log format. Maybe it's time for new log format with more features and bigger room for names and arrays support (some topics already have arrays)? Also IMU, IMU1, IMU2 looks not good. It would be better to have some topic indexes for such cases.

LorenzMeier, 2015-07-13:
I think we indeed might need to revise the log format in the process and build support for it into FlightPlot and pymavlink. I agree with Anton that the current format is limiting.

DrTon, 2015-07-22:
Support for the described format is already in FlightPlot, but not in master yet, let me know if you want to play with it.

LorenzMeier, 2016-05-22:
Finished!

closed 2016-05-22 by LorenzMeier