PX4/PX4-Autopilot #2489: FMUv1 randomly erazes params

bansiesta, 2015-06-28:
Every now and then, it seems that all params are wiped from the FMUv1 EEPROM:

```
mtd: failed to initialize EEPROM driver
param: selected parameter default file /fs/microsd/params
[param] Loaded: /fs/microsd/params
[rgbled] on I2C bus 3 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 0:   SYS_AUTOSTART: match
[i] No autostart
[PX4IO_i2c] on I2C bus 2 at 0x1a (bus: 100 KHz, max: 400 KHz)
px4io: CRCs match
dataman: Unknown restart, data manager file '/fs/microsd/dataman' size is 101504 bytes
[MS5611_I2C] on I2C bus 2 at 0x76 (bus: 100 KHz, max: 400 KHz)
ms5611: no device on bus 2
<adc> init done
[MPU6000] on SPI bus 1 at 3 (1000 KHz)
[L3GD20] on SPI bus 1 at 1 (11000 KHz)
hmc5883: no device on bus 2
[HMC5883_I2C] on I2C bus 2 at 0x1e (bus: 100 KHz, max: 400 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
nsh: px4flow: command not found
telem> PREFLIGHT FAIL: MAG #0 UNCALIBRATED
telem> PREFLIGHT FAIL: ACCEL #0 UNCALIBRATED
telem> PREFLIGHT FAIL: GYRO #0 UNCALIBRATED
telem> PREFLIGHT FAIL: GYRO #1 UNCALIBRATED
telem> PREFLIGHT FAIL: AIRSPEED SENSOR MISSING
```

Any ideas why this would happen?

sjwilks, 2015-06-28:
I had this happen once just a few days ago too on a V1 board. Thought it
might just have been a one off but it looks like it isn't the case.

mhkabir, 2015-06-30:
I have seen this on FMU v2 as well 2-3 times. Just a reboot, no config changes, etc. yet the params are all reset. Initially dismissed as a one-off, but this keeps happening.
Couldn't find a way to reproduce it reliably.

TSC21, 2015-07-01:
I also confirm this issue on FMUv2. It sometimes happens when I load new firmware to the board. It's difficult to reproduce as it doesn't happen within a fixed pattern, so I'm not able to figure when it happens or not. But all sensor callibrations, for example, disappear.

LorenzMeier, 2015-11-17:
All reports I've seen is when people reboot while params are written. There is nothing we can do about this other than telling people to use their system correctly (you also don't shut down your computer while Powerpoint is still saving your slide deck). Closing this.

closed 2015-11-17 by LorenzMeier