PX4/PX4-Autopilot #2490: Bad parameter meta data

DonLakeFlyer, 2015-06-28:
I'm in the process of completing a pull in QGC which validates parameter value modification against the metadata. While doing that I've noticed quite a bit of validation meta-data which is incorrect. Here is a log from QGC which now checks the following things in the meta data:
- Values can be correctly converted to the parameter type
- Default value is within specified min/max values

I could used some help fixing these up. I may even make sense to fix these prior to the Release. The reason being as soon a I merge in the parameter validation pull on the QGC side it's going to yell at you if your value if out of bounds. That's going to get annoying if the only reason you value change is out of bounds is because the min/max in the meta dat is wrong. Especially if the default value is out of bounds to start with!

In reality the min/max values need to full review for all parameters to make sure they are correct. These are just the ones which are completely out of whack.

You can disregard the blank default value errors. These are caused by the python parser not being able to do the right thing in some cases.

PX4ParameterLoaderLog: Invalid default value, name: "COM_EF_TIME"  type: 6  default: QVariant(float, 10)  error: "Value must be within 0 and 7"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_DLL_CH_LAT"  type: 5  default: QVariant(int, -266072120)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_AH_LAT"  type: 5  default: QVariant(int, -265847810)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "MAV_TYPE"  type: "INT32"  default: "MAV_TYPE_FIXED_WING"  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ROLL_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ROLLRATE_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ROLLRATE_I"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ROLLRATE_D"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_PITCH_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_PITCHRATE_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_PITCHRATE_I"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_PITCHRATE_D"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAW_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAWRATE_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAWRATE_I"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAWRATE_D"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAW_FF"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_YAWRATE_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ACRO_R_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ACRO_P_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MP_ACRO_Y_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_MAN_R_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_MAN_P_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_MAN_Y_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid max value, name: "MPC_Z_VEL_MAX"  type: 6  max: "8 m/s"  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_THR_MIN"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_THR_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_VEL_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_VEL_I"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_VEL_D"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_VEL_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_Z_FF"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_VEL_P"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_VEL_I"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_VEL_D"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_VEL_MAX"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_XY_FF"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_TILTMAX_AIR"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_TILTMAX_LND"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "MPP_LAND_SPEED"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "PE_MAG_NOISE"  type: 6  default: QVariant(float, 0.05)  error: "Value must be within 0.1 and 10"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
[... cut ...]
PX4ParameterLoaderLog: Invalid default value, name: "SUB_INTERV"  type: "INT32"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "SUB_TESTF"  type: "FLOAT"  default: ""  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "VT_MOT_COUNT"  type: 5  default: QVariant(int, 0)  error: "Value must be within 1 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "VT_PROP_EFF"  type: 6  default: QVariant(float, 0)  error: "Value must be within 0.5 and 0.9"

DonLakeFlyer, 2015-06-28:
@LorenzMeier Note the comment on whether this should go into the Release or not.

DonLakeFlyer, 2015-06-28:
I changed the logging to not report blank default values:

PX4ParameterLoaderLog: Invalid default value, name: "COM_EF_TIME"  type: 6  default: QVariant(float, 10)  error: "Value must be within 0 and 7"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_DLL_CH_LAT"  type: 5  default: QVariant(int, -266072120)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_AH_LAT"  type: 5  default: QVariant(int, -265847810)  error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "MAV_TYPE"  type: "INT32"  default: "MAV_TYPE_FIXED_WING"  error: "Invalid number"
PX4ParameterLoaderLog: Invalid max value, name: "MPC_Z_VEL_MAX"  type: 6  max: "8 m/s"  error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "PE_MAG_NOISE"  type: 6  default: QVariant(float, 0.05)  error: "Value must be within 0.1 and 10"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_TRIM"  type: 6  default: QVariant(float, 1500)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_MIN"  type: 6  default: QVariant(float, 1000)  error: "Value must be within 0 and 100"
[... cut ...]
PX4ParameterLoaderLog: Invalid default value, name: "RC18_MAX"  type: 6  default: QVariant(float, 2000)  error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC_FAILS_THR"  type: 5  default: QVariant(int, 0)  error: "Value must be within 800 and 2200"
PX4ParameterLoaderLog: Invalid default value, name: "VT_MOT_COUNT"  type: 5  default: QVariant(int, 0)  error: "Value must be within 1 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "VT_PROP_EFF"  type: 6  default: QVariant(float, 0)  error: "Value must be within 0.5 and 0.9"

LorenzMeier, 2015-06-29:
Let me have a pass

LorenzMeier, 2015-06-30:
Fixed in stable / beta, which will also be merged to master. Closing.

closed 2015-06-30 by LorenzMeier