PX4/PX4-Autopilot #2494: Can't enter the BIND mode with some DSMX Remote Receiver(SPM9645)

cctsao1008, 2015-06-29:
I just bought five DSMX Remote Receivers (SPM9645) from online store, All of them cannot get into BIND mode using the Qgroundcontrol, after change the pulse delay from 25us to 120us,the BIND mode works for each receiver.

cctsao1008, 2015-06-29:
Reference to the Baseflight & Cleanflight

/* spektrumBind function ported from Baseflight. It's used to bind satellite receiver to TX.

Function must be called immediately after startup so that we don't miss satellite bind window.
Known parameters. Tested with DSMX satellite and DX8 radio. Framerate (11ms or 22ms) must be selected from TX.
9 = DSMX 11ms / DSMX 22ms
5 = DSM2 11ms 2048 / DSM2 22ms 1024
*/
void spektrumBind(rxConfig_t *rxConfig)
{
int i;
if (!spekShouldBind(rxConfig->spektrum_sat_bind)) {
return;
}

LED1_ON;

gpio_config_t cfg = {
BIND_PIN,
Mode_Out_OD,
Speed_2MHz
};
gpioInit(BIND_PORT, &cfg);

// RX line, set high
digitalHi(BIND_PORT, BIND_PIN);

// Bind window is around 20-140ms after powerup
delay(60);
LED1_OFF;

for (i = 0; i < rxConfig->spektrum_sat_bind; i++) {

LED0_OFF;
LED2_OFF;
// RX line, drive low for 120us
digitalLo(BIND_PORT, BIND_PIN);
delayMicroseconds(120);

LED0_ON;
LED2_ON;
// RX line, drive high for 120us
digitalHi(BIND_PORT, BIND_PIN);
delayMicroseconds(120);
}

#ifndef HARDWARE_BIND_PLUG
// If we came here as a result of hard reset (power up, with spektrum_sat_bind set), then reset it back to zero and write config
// Don't reset if hardware bind plug is present
if (!isMPUSoftReset()) {
rxConfig->spektrum_sat_bind = 0;
saveConfigAndNotify();
}
#endif

}
#endif

cctsao1008, 2015-06-29:
Reference to the OpenPilot

/* Try to bind DSMx satellite using specified number of pulses */
static void PIOS_DSM_Bind(struct pios_dsm_dev *dsm_dev, uint8_t bind)
{
const struct pios_dsm_cfg *cfg = dsm_dev->cfg;

GPIO_InitTypeDef GPIO_InitStructure = cfg->bind.init;

GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_UP;

/* just to limit bind pulses */
if (bind > 10) {
    bind = 10;
}

GPIO_Init(cfg->bind.gpio, &cfg->bind.init);

/* RX line, set high */
GPIO_SetBits(cfg->bind.gpio, cfg->bind.init.GPIO_Pin);

/* Wait until the bind window opens. */
while (PIOS_DELAY_GetuS() < DSM_BIND_MIN_DELAY_US) {
    ;
}

for (int i = 0; i < bind; i++) {
    /* RX line, drive low for 120us */
    GPIO_ResetBits(cfg->bind.gpio, cfg->bind.init.GPIO_Pin);
    PIOS_DELAY_WaituS(120);
    /* RX line, drive high for 120us */
    GPIO_SetBits(cfg->bind.gpio, cfg->bind.init.GPIO_Pin);
    PIOS_DELAY_WaituS(120);
}
/* RX line, set input and wait for data */
GPIO_Init(cfg->bind.gpio, &GPIO_InitStructure);
}

cctsao1008, 2015-06-29:
This is my pull request..
https://github.com/PX4/Firmware/pull/2493

LorenzMeier, 2015-06-29:
Fixed, thanks!

closed 2015-06-29 by LorenzMeier