PX4/PX4-Autopilot #2495: Set correct defaults for all params

LorenzMeier, 2015-06-29:
@DonLakeFlyer Tracking this here to fix up meta data.

I changed the logging to not report blank default values:

PX4ParameterLoaderLog: Invalid default value, name: "COM_EF_TIME" type: 6 default: QVariant(float, 10) error: "Value must be within 0 and 7"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_DLL_CH_LAT" type: 5 default: QVariant(int, -266072120) error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "NAV_AH_LAT" type: 5 default: QVariant(int, -265847810) error: "Value must be within 0 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "MAV_TYPE" type: "INT32" default: "MAV_TYPE_FIXED_WING" error: "Invalid number"
PX4ParameterLoaderLog: Invalid max value, name: "MPC_Z_VEL_MAX" type: 6 max: "8 m/s" error: "Invalid number"
PX4ParameterLoaderLog: Invalid default value, name: "PE_MAG_NOISE" type: 6 default: QVariant(float, 0.05) error: "Value must be within 0.1 and 10"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC3_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC4_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC5_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC6_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC7_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC8_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC9_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC10_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC11_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC12_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC13_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC14_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC15_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC16_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC17_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC18_MIN" type: 6 default: QVariant(float, 1000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC18_TRIM" type: 6 default: QVariant(float, 1500) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC18_MAX" type: 6 default: QVariant(float, 2000) error: "Value must be within 0 and 100"
PX4ParameterLoaderLog: Invalid default value, name: "RC_FAILS_THR" type: 5 default: QVariant(int, 0) error: "Value must be within 800 and 2200"
PX4ParameterLoaderLog: Invalid default value, name: "VT_MOT_COUNT" type: 5 default: QVariant(int, 0) error: "Value must be within 1 and 2147483647"
PX4ParameterLoaderLog: Invalid default value, name: "VT_PROP_EFF" type: 6 default: QVariant(float, 0) error: "Value must be within 0.5 and 0.9"

DonLakeFlyer, 2015-06-29:
Thanks Lorenz. After this is cleaned up I can change the source parser to validate these on the firmware end instead of QGC. Then this turns into build breaks on the firmware side when things are wrong.

LorenzMeier, 2015-06-30:
Duplicate of #2490

closed 2015-06-30 by LorenzMeier