PX4/PX4-Autopilot #2521: POSIX: Launching thread should not return immediately

LorenzMeier, 2015-07-02:
In this line the thread we're launching is returning immediately. Instead what we want is the same behaviour as from a command line application, where the execution blocks until it is returning:
https://github.com/PX4/Firmware/blob/master/src/platforms/posix/main.cpp#L71

The usleep below could then also be removed.

mcharleb, 2015-07-02:
The application runs the "start" or other command passed through it via argv. The start handling usually spawns a thread to run the app, and then the main returns while the other thread continues runs asynchronously.

What is the main thread (also the shell thread) supposed to wait for?

LorenzMeier, 2015-07-02:
The issue is that your shell is not behaving like a shell. A normal shell blocks until the executed program ends. Because its not behaving as it should (and our wait logic hence can't operate as it should) we run into a *a lot* of race issues.
So what we really need is to wait until the thread returns. Because the thread you're launching is only the shell / start handler, and spawns another thread which does the actual execution.

mcharleb, 2015-07-02:
I'll have a look.

mcharleb, 2015-07-02:
I am still not convinced your conclusions are correct. The flow for NuttX and for POSIX seem the same to me:

Nuttx shell runs dataman start which calls dataman_main(argc, argv)
That calls start() which sets a semaphore and spawns a task and waits for the waits for the task to initialize via the semaphore, and then returns.

The flow is the same for POSIX:

mainapp shell runs "dataman start" which calls dataman_main(argc, argv)
That calls start() which sets a semaphore and spawns a task and waits for the task to initialize via the semaphore, and then returns.

The call does not return immediately unless the start function is coded to do so.

mcharleb, 2015-07-02:
A "normal shell" won't block on a task that spawns new tasks, and then exits.

LorenzMeier, 2015-07-03:
That's all fair, but the code is not doing what you think it's doing. Here is the output from starting the simulator app, which has a while() loop which only returns once it has the first data:
```
Running: simulator
arg 0 simulator
arg 1 start
arg 2 -s
Returning: simulator
INFO  Simulator started
WARN  1          failed to open /dev/ttyACM0 (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 379)
WARN  113        failed to send streaming command to /dev/ttyACM0 (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 387)
WARN  191        Waiting for initial data on UDP.. Please connect the simulator first. (file /home/drone/src/Firmware/src/modules/simulator/simulator_mavlink.cpp line 403)
```

Your start handler table, which I already linked above, is non-blocking. And that's the core issue which needs to be fixed. Its currently *not* behaving like a shell and the complete system startup sequencing is messed up.
https://github.com/PX4/Firmware/blob/master/src/platforms/posix/main.cpp#L71

LorenzMeier, 2015-07-03:
I have pushed this state to master, so if you just run ```make posix && make sitl_quad``` you will get the output I shared here.

LorenzMeier, 2015-07-04:
@mcharleb My apologies - I did try to do proper testing, but I screwed up. My bad testing led to the wrong conclusions. I've fixed everything now. The required fixes where 1) to wait for the sim and 2) to reset the monotonic system time to 0 when booting. I want to long-term identify the regression enforcing this, but the fact that the monotonic clock *has to be* at zero is part of the system design spec and so not really a real issue or regression.

closed 2015-07-04 by LorenzMeier