PX4/PX4-Autopilot #2523: Battery Remaining Estimate Voltage flawed

orangelynx, 2015-07-02:
Hi everyone,

for background see: https://groups.google.com/forum/#!topic/px4users/Aqq20MLNr-E

after experiencing some issues with jumping "battery remaining"-values, I took a look at  the source and quickly found the (in my opinion) problem in the commander_helper.cpp function battery_remaining_estimate_voltage.

The voltage drop is only taken into consideration for the maximum battery voltage, not the minimum or the current measurement. To correctly take internal resistance into account, one should apply it to all voltage values.

Example which is causing the problem:
Say you have a 3S Battery. Maximum Cell Voltage is 4.2V, minimum is set to 3.6V. Due to a high internal resistance, V_DROP is 1V.

    /* remaining charge estimate based on voltage and internal resistance (drop under load) */
	float bat_v_full_dynamic = bat_v_full - (bat_v_load_drop * throttle_normalized);
	float remaining_voltage = (voltage - (bat_n_cells * bat_v_empty)) / (bat_n_cells * (bat_v_full_dynamic - bat_v_empty));

Playing it through, this will lead to bat_v_full_dynamic = -0.4V causing sign issues in the calculation for remaining_voltage.

Solution: At least  to my knowledge, it's perfectly possible for the battery to drop under load to BAT_V_EMPTY - BAT_V_LOAD_DROP per cell, and this should be properly taken into account. A quick thought would  be
    float bat_v_empty_dynamic = bat_v_empty - (bat_v_load_drop * throttle_normalized);
    float remaining_voltage = (voltage - (bat_n_cells * bat_v_empty_dynamic)) / (bat_n_cells * (bat_v_full - bat_v_empty));

I would create a pull request but since I have little knowledge about the whole project, I thought it'd be best to just post it as an issue.

LorenzMeier, 2015-07-03:
You're right about your observation. Thanks!

closed 2015-07-03 by LorenzMeier

orangelynx, 2015-07-03:
Dear Lorenz,

thanks for the fix, however please keep in mind that setting the V_DROP parameter incorrectly can now lead to critical discharge of the battery without being warned. I don't know where else in the project this might have an effect on, just keep it in mind.

I'll also post a suggestion to QGroundControl to add a warning to that parameter.

commit e23459e8 (2015-07-03): Commander: Fix dynamic battery scaling, proposed by @orangelynx. Fixes #2523.
commit 9e223f0c (2015-07-03): Commander: Fix dynamic battery scaling, proposed by @orangelynx. Fixes #2523.