PX4/PX4-Autopilot #2526: Need urgent help , Plz.. Why I can't calibrate the joystick

AsMoh, 2015-07-03:
Hi,
We tried to fly our quadcopter using joystick, and it worked. Then, we wanted to try to use turnigy transmitter and turnigy 9x8c receiver, but it did not work maybe because we connected the receiver to the pixhawk by PPM encoder , and the turnigy 9x8c receiver gives PPM output. I mean the connection was wrong.

Therefore, we decided to continue using the joystick in our graduation project.
When we tried to arm the quadcopter, an error appears and it says "Prearm rc not calibrated" although the joystick was calibrated and it was working fine. 
We tried to calibrate the joystick again, but the green bars are not showing. The mission planner detects the joystick, but no green bars show when we move the buttons and the slider dedicated to the pitch, roll, throttle and rudder, I mean the important channels. Only when we press the other buttons( the ones for changing modes, arm , disarm..) , the green color shows.
We re-installed the mission planner, but we are getting the same problem. We changed the joystick to another compatible joystick , and it did not work. No green bars although mission planner detects it. 
Before buying this pixhawk, we had apm2.6 and we got the same problem. But ,in that time, we were able to calibrate all channels except CH7. We thought that maybe some pins got burned because we removed the GPS and the apm2.6 was powered on. Thus, we bought a pixhawk.
But now , we just tried this turnigy transmitter and receiver, we did not remove any wires.
Can you help us to know why we can't calibrate the joystick, please? We don't have time to buy new pixhawk. Thank you in advance

TSC21, 2015-07-03:
Are you using PX4 Firmware or Arducopter's?

AsMoh, 2015-07-03:
I am using Arducopter 3.2.1

TSC21, 2015-07-03:
So you should forward your question to Ardupilot repo: https://github.com/diydrones/ardupilot. This is PX4 Firmware only. Please close the issue and place this on the Adupilot issue list. And good luck ;)

AsMoh, 2015-07-03:
Okk.. Thank you

closed 2015-07-03 by AsMoh