PX4/PX4-Autopilot #2532: [TECS] Performance Altitude Control

boosfelm, 2015-07-05:
In general [altitude tracking!] (http://dash.oznet.ch/view/HMD4a9ZMsYM9hJbBZtufvS) looks pretty good, however in turns (600s to 1000s) the plane looses about 5m altitude. Increasing ff roll to throttle (RLL2THR) to 20 did not help much, the plane obviously speeds up but does not provide enough lift. This has to be compensated by a Roll to Pitch ff term. Is this already done by someone or should we do it and send a pr? I believe this is highly relevant for most "less conventional" airplanes.

[image]

SimonWilks, 2015-07-05:
Have you seen/tried FW_P_ROLLFF?

boosfelm, 2015-07-05:
Hi @SimonWilks 
Thanks for your comment. I grepped for FW_P_ROLLFF and have found it in fw_att_control, where I didn't expect it. We are flying our own quternion-based attitude module and interfaced to fw_pos_control. Do you think this should even be in the attitude module? 
I had a look at the code but I can't figure out the logic at a first glance. How is ff pitch for rolling motions performed?
Thanks for your help and have a nice sunny day, Max

boosfelm, 2015-07-05:
Trying to explain my thought about sw architecture in more detail. 
In a banking maneuver we loose lift and thus descend, however pitch is not directly affected (although we want ff pitch to compensate for it). Hence I believe a tuning parameter should be rather part of the position controller than of the attitude controller.

LorenzMeier, 2015-11-01:
Closing this in favour of #2980.

closed 2015-11-01 by LorenzMeier