PX4/PX4-Autopilot #2533: POSIX: Navigator: px4_poll() / hrt related issues

LorenzMeier, 2015-07-05:
It turns out that the navigator app, which subscribes to 8 topics out of which two are rate limited, can get into a weird state resulting in this error message when debugging logic is enabled:
```
WARN  4936378    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 5004268    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  5137807    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 5204467    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  5339233    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 5404616    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  5540869    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 5604773    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  5741445    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 5804983    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  5942247    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 6005594    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
WARN  6143076    0XDAEB7700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
ERROR 6205776    0XD8AA5700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
```

LorenzMeier, 2015-07-05:
Another run ends up in a segfault, even with orb_set_interval() calls removed:
```
Running: mavlink
arg 0 mavlink
arg 1 start
arg 2 -u
arg 3 14556
arg 4 -r
arg 5 60000
WARN  222105     0XC4E51F00 starting task mavlink_if0 (file /Firmware/src/platforms/posix/px4_layer/px4_posix_tasks.cpp line 145)
WARN  222894     0XD2612700 mode: 0, data rate: 60000 B/s on udp port 14556 (file /Firmware/src/modules/mavlink/mavlink_main.cpp line 1487)
INFO  Setting up UDP w/port 14556

WARN  371623     0XD1208700 poll: bad fd state (file /Firmware/src/drivers/device/vdev.cpp line 458)
WARN  371907     0XD1208700 timed out (file /Firmware/src/modules/navigator/navigator_main.cpp line 340)
WARN  372428     0XD1208700 offboard mission update failed (file /Firmware/src/modules/navigator/mission.cpp line 268)
WARN  373017     0XD1208700 ERROR: invalid mission state (file /Firmware/src/modules/navigator/mission.cpp line 748)
Returning: mavlink
Enter a command and its args:
INFO  ToneAlarm::start_note 8522
ERROR 990181     0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 990959     0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 992475     0XD0A04700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 997986     0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1045781    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1049879    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1054299    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1058538    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1063115    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1067297    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1071441    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1075551    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1079804    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1084021    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1088267    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1092680    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1096751    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1099692    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1101347    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1105522    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1110936    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1115195    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
INFO  ToneAlarm::start_note 6384
ERROR 1116276    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1119569    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1128329    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1132606    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1136831    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1140986    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1145154    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1149306    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 1153488    0XC6C58700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
Tools/sitl_run.sh: line 5: 18019 Segmentation fault      (core dumped) ./mainapp ../../$1
make: *** [sitl_quad] Error 139
```

LorenzMeier, 2015-07-05:
Another trace without crash, but this time many involved threads:
```
ERROR 275722082  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275722161  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275722703  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275746587  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275767249  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275772305  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275772388  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275772423  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275772524  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275793536  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275814084  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275822593  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275822663  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275822810  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275843063  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275863964  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275872856  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275872975  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275873103  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275888003  0X65007700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275894071  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275914592  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275923301  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275923402  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275923557  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275943849  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275964143  0X66612700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275973746  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275973830  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275973862  0X6560A700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275973997  0X66E16700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 275984293  0X5F8C0700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
ERROR 276004317  0X5F8C0700 hrt_call_internal interval too short: 0 (file /Firmware/src/platforms/posix/px4_layer/drv_hrt.c line 358)
```

LorenzMeier, 2015-07-05:
@mcharleb This looks like it might be worth looking into.

LorenzMeier, 2015-07-05:
More testing feedback:
```
----------------------------------
Running: mixer
WARN  3261340    offboard mission update failed, handle: 72 (file /Firmware/src/modules/navigator/mission.cpp line 268)
WARN  3261601    ERROR: invalid mission state (file /Firmware/src/modules/navigator/mission.cpp line 748)
<hilservo> lock

```

In general a lot of variability of the results, very much looking like a race.

LorenzMeier, 2015-07-08:
@mcharleb This commit fixes it:
fc3a85311d377768f8ae2a2761c23c346115e824

So my current conclusion is that orb_advertise() is not thread safe and that its the root cause.

mcharleb, 2015-07-10:
I haven't had a chance to investigate it yet. I'll have a look.

LorenzMeier, 2015-11-25:
Fixed.

closed 2015-11-25 by LorenzMeier