PX4/PX4-Autopilot #2534: Request data stream sets wrong rates

AndreasAntener, 2015-07-05:
@LorenzMeier 
In `handle_message_request_data_stream` we convert the received rate (I believe it's meant to be a rate, see below) into an interval (https://github.com/PX4/Firmware/blob/bc5cf50f1a2e3c6113204c48258835d9098b5b0b/src/modules/mavlink/mavlink_receiver.cpp#L1166). But the later invoked method to configure the stream in `mavlink_main` does it again which leads to huge rates. This might be bug in `handle_message_request_data_stream`, which I can fix as soon as I know what the message really meant to specify.

In `REQUEST_DATA_STREAM`, the comment says it specifies an interval, but the parameter name suggests it's a rate.
https://github.com/mavlink/mavlink/blob/378cd337301b3a6b944fedc46b4fcd89e16da42c/message_definitions/v1.0/common.xml#L2095

What is it? Rates seem to make more sense to me, so I assume the comment lies (what a surprise that would be ;)). Or is it rates * 1000 so we can specify slower than 1Hz?

AndreasAntener, 2015-07-05:
The only usage example I currently have is the ArduCAM OSD which requests rates of 5 or 2.

LorenzMeier, 2015-07-08:
It needs a completely new API and message spec. What kind of use case do you have / what things would you like to see in the API?

AndreasAntener, 2015-07-09:
I'm don't want to invent any use-cases here, so I'm trying to keep the OSD or our onboard computers in mind and what we do now:
The component connects via MAVLink. The FCS doesn't have to know about this component, only that it's connected, an initial link would only stream heartbeats. The component then requests FCS capabilities, decides which messages might be supported off that information, then it requests streams of those messages.

So the API would define:
- request_stream_of_messages(message_id, rate)

For more link management (already inventing functionality here I don't need yet..):
- set_rate_of_message_stream(message_id, rate) // rate == 0 would mean stop

I'm explicitly talking about *messages* here. The concept of abstract *streams* that seems to be the idea behind the current API, which can hide multiple different messages for different FCS, doesn't seem to be necessary in that use-case. Thoughts:
- would it even be possible to use the streams like that? Technically we compile against one specific dialect anyway, and conceptually different FCS could use the same message IDs for different messages which would make the parsing difficult.
- most of the messages used are in common anyway, an effort to bring frequently used messages to the common format is better than having abstract streams

LorenzMeier, 2015-07-09:
Yes, that was pretty much along the lines of what I was thinking. I just need to implement the API then and more importantly apps like QGC need to learn how to use it.

AndreasAntener, 2015-07-09:
Ok. It seems my second method is not actually necessary, could all be done with one, assuming it really stays that simple:
- configure_stream_of_messages(message_id, rate) // rate == 0 would mean stop

LorenzMeier, 2015-08-11:
New API is implemented in the new MESSAGE_INTERVAL message and the MAV_CMD_SET_MESSAGE_INTERVAL and MAV_CMD_GET_MESSAGE_INTERVAL. Its not implemented on the PX4 side, but a pull request would be most welcome.

LorenzMeier, 2015-12-02:
Moving to next release. Will do this once it becomes wanted.

LorenzMeier, 2016-05-22:
Implemented, closing.

AndreasAntener, 2016-05-25:
I don't think this is implemented

LorenzMeier, 2016-07-29:
It definitively is. Closing.

closed 2016-07-29 by LorenzMeier