PX4/PX4-Autopilot #2541: Missing iris_*_world_with_joy.launch

weiweikong, 2015-07-08:
Hi,
     just tested PX4 ROS SITL Setup, following the guide from https://pixhawk.org/dev/ros/sitl#joystick_configuration. But failed when I tried
```
roslaunch px4 gazebo_iris_house_world.launch
```
or 
```
roslaunch px4 gazebo_iris_outdoor_world.launch
```

Checking the launch file `Firmware/launch/gazebo_iris_outdoor_world.launch` and `Firmware/launch/gazebo_iris_house_world.launch`, it seems two other launch files, `iris_outdoor_world_with_joy.launch` and `iris_outdoor_world_with_joy.launch`, are missing.

LorenzMeier, 2015-07-08:
Thanks for reporting. I'm currently reworking this and will have something more wrapped up tomorrow.

kPanesar, 2015-11-22:
Any updates on this issue? The repository still seems to be missing the files.

weiweikong, 2015-11-23:
@kPanesar you could try other launch file or build a new world for it.

kPanesar, 2015-11-23:
@weiweikong True, I was just wondering if it there were any updates as it has been a few months. Thanks!

LorenzMeier, 2016-05-22:
Outdated.

closed 2016-05-22 by LorenzMeier