PX4/PX4-Autopilot #2551: GPS driver keeps publishing even when GPS isn't connected

mhkabir, 2015-07-10:
When no GPS is connected, the driver still keeps spamming the topic with empty messages. Driver in Ashtech mode when this occurs.

Running a UAVCAN-interfaced GPS in parallel then causes weird issues.

@kd0aij Also brought this up.

LorenzMeier, 2015-07-10:
Indeed, we need to fix that. I don't recall why this would be necessary.

LorenzMeier, 2015-07-10:
@Zefz I think you introduced this, but I don't recall what the use case was.

Zefz, 2015-07-15:
@LorenzMeier My GPS check only runs if you have the circuit breaker is disabled (CBRK_GPSFAIL). It is on by default so unless you explicitly requested it, the Preflight GPS check will not run.

My preflight check is here: https://github.com/PX4/Firmware/blob/master/src/modules/commander/PreflightCheck.cpp#L277-L305

This is not the same as messages coming from the GPS driver as @mhkabir is mentioning?

@LorenzMeier Use case: Detect faulty GPS wiring, broken GPS receiver or GPS receivers that are placed too close to the RC receiver (which will kill all reception because of the noise). Without this check the system does not report anything wrong - it will simply never get a fix :)

LorenzMeier, 2015-07-17:
Its the ashtech driver which returns success for no reason:
https://github.com/PX4/Firmware/blob/master/src/drivers/gps/ashtech.cpp#L647-L661

LorenzMeier, 2015-07-17:
Closed in d9f0baf on beta.

closed 2015-07-17 by LorenzMeier